drtyhlpr/rpi23-gen-image #10: Detached screen terminals get terminated when you log out

fangfufu, 2016-01-02:
GNU Screen works in a weird way in the image generated using your script - detached screen terminals get terminated when you log out. 

To test this, you need to connect to your Raspberry Pi 2 via SSH, and then do the following: 

    $ ssh $IP_ADDRESS
    $ screen
    # type in ctrl +d to detach your screen session
    $ screen -r
    # re-attachment works correctly
    # type in ctrl +d to detach your screen session
    $ exit # we are logging out from 
    #let's log back in via SSH
    ssh $IP_ADDRESS
    $ screen -r
    There is no screen to be resumed.

I have no idea how to fix this, apologies if I haven't explained myself properly. I have never encountered this problem before, I have no idea what to type in to Google in order to search for solutions. 

This is how I generated my image:

    APT_SERVER="http.debian.net" HOSTNAME="vanguard" PASSWORD="afvzsdcx" DEFLOCAL="en_GB.UTF-8" TIMEZONE="Europe/London" ENABLE_CONSOLE=true ENABLE_IPV6=true ENABLE_SSHD=true ENABLE_SOUND=false ENABLE_MINGPU=true ENABLE_DBUS=false ENABLE_XORG=false ENABLE_UBOOT=true ./rpi2-gen-image.sh

I had to install dbus after generating the image by running:

    sudo apt-get install dbus

drtyhlpr, 2016-01-04:
thanks for your feedback - i need to investigate this - hm haven't tested screen yet.

ENABLE_DBUS=false was set on your command line - maybe by mistake - this is why u need to post-install dbus manually.

fangfufu, 2016-01-04:
Yes, I was informed by OpenVPN that I needed dbus - I thought I didn't need it. I thought providing this extra bit of information might be of some use. 

By the way, I detached the terminal by pressing Ctrl + A, then d. I typed it wrong.

drtyhlpr, 2016-01-04:
@fangfufu 
all my rpi2s/sdcards are currently busy - I need more sdcards ;] / so I need some more days to test and verify your problem.

here is a post that sounds similar to the problem you described: http://serverfault.com/questions/660728/screen-r-not-working-with-ssh

if you have time please send me the output of the following commands:
screen -ls
(and or)
ls -laR /var/run/screen/

fangfufu, 2016-01-05:
I connected to two SSH sessions, these commands were performed in one session that was kept on. 

Before exiting the other SSH session: 

    fangfufu@vanguard:~$ screen -ls
    There is a screen on:
            2169.pts-2.vanguard     (05/01/16 01:30:48)     (Detached)
    1 Socket in /var/run/screen/S-fangfufu.

    fangfufu@vanguard:~$ ls -laR /var/run/screen/
    /var/run/screen/:
    total 0
    drwxrwxr-x  3 root     utmp      60 Jan  5 01:30 .
    drwxr-xr-x 16 root     root     500 Jan  3 00:35 ..
    drwx------  2 fangfufu fangfufu  60 Jan  5 01:30 S-fangfufu

    /var/run/screen/S-fangfufu:
    total 0
    drwx------ 2 fangfufu fangfufu 60 Jan  5 01:30 .
    drwxrwxr-x 3 root     utmp     60 Jan  5 01:30 ..
    prw------- 1 fangfufu fangfufu  0 Jan  5 01:30 2169.pts-2.vanguard
    fangfufu@vanguard:~$

After exiting the SSH session: 

    fangfufu@vanguard:~$ screen -ls
    No Sockets found in /var/run/screen/S-fangfufu.

    fangfufu@vanguard:~$ ls -laR /var/run/screen/
    /var/run/screen/:
    total 0
    drwxrwxr-x  3 root     utmp      60 Jan  5 01:30 .
    drwxr-xr-x 16 root     root     500 Jan  3 00:35 ..
    drwx------  2 fangfufu fangfufu  40 Jan  5 01:33 S-fangfufu

    /var/run/screen/S-fangfufu:
    total 0
    drwx------ 2 fangfufu fangfufu 40 Jan  5 01:33 .
    drwxrwxr-x 3 root     utmp     60 Jan  5 01:30 ..
    fangfufu@vanguard:~$

drtyhlpr, 2016-01-05:
ok - my first impression is that this seems not to be a problem related to the image creation process - but I may be wrong - I will try to solve your issue this weekend....wired problem...

here is another post - that really ;] sounds like the problem you described:
http://superuser.com/questions/58525/how-do-i-reconnect-to-a-lost-screen-detached-missing-socket

fangfufu, 2016-01-05:
I have previous debootstrap my own Debian image before, not using your script, using the instructions from:
https://wiki.debian.org/MatthiasSchmitz#Building_own_Debian_image_for_Raspberry_Pi

It worked fine for me. I didn't have this strange screen problem. I have a filling that some permissions might not have been set correctly / signals aren't passing through correctly. In the image created using your script, ping doesn't work natively. I had to give the binary the +S flag.

drtyhlpr, 2016-01-07:
hm I understand your point. but my point is that I am just using the standard installer utilities for a debian linux to build an image. Times have changed and I feel like Linux became Windows in the way that a (well tested) default install of linux distros uses GBs of data and the user has to remove all the stuff he possibly does not need. if you check the dependency-relation-tree it is still hard to get a small final OS with all the tools and GUIs you maybe like even if you use the created images by this script - well thats my point of view. (summary: this text somehow says that it's debian's fault / because I am just using their minimal installer fe. type 'minbase' does not work without minimal mods / so I conclude that there may be small errors or problems if you don't use tasksel install (like ping / well I am sure I used ping but I will test that too).

I will try to help you, but you seem to have advanced linux/debian skills (like your own bootstrap instructions etc.) - so I am NOT sure if I can. so fe. I am sure you already used strace etc

fangfufu, 2016-01-07:
Ah, no worries. I am not that good with Linux. I haven't used strace. I know how to copy and paste other people's command, that's about it. 

Your script is wonderful, by the way. Keep up with the good work. :)

drtyhlpr, 2016-01-07:
thanks - but of course I do not want to offer a script without working screen and ping / which are very essential utilities :)

I am happy you like it - I just created it to save people time / maybe bring some people back to smaller OS sizes (and finally to RIOT OS... hehe) / work on their project instead of messing with the bootstrapping etc.

fangfufu, 2016-01-07:
I think ping's problem is easy to solve. You are sort of right about tasksel. I have occasionally created images that required me to manually give the +s flag to ping. But I am absolutely clueless about screen though. 

I think running Debian is simply better than running Raspbian. You get to use all the random repositories that people have put on the Internet, especially deb-multimedia.

drtyhlpr, 2016-01-07:
"In any event ping won't be usable until netbase has been configured, rather than simply unpacked. dpkg --pending --configure will fix that."

`dpkg --pending --configure` heh first time I hear about that command ... hmmm (added to list of things to test to fix described problems)

fangfufu, 2016-01-08:
Where did you get this piece of information from? It is kind of useful
for me to learn about these too. :)

parobalth, 2016-01-10:
This problem existed also with my image. It is wicked to google for a solution because all you get are solutions to other screen related problems. Somehow the screen process gets killed when you logout. My working solution was to reinstall the packages 'openssh-client' and 'openssh-server'. Then I restarted the ssh server with 'sudo service sshd restart'. Screen now works as expected. Please tell me if this solution fixes the problem for you too.

parobalth, 2016-01-19:
A persistent workaround is "sudo apt-get -y purge openssh-server && sudo apt-get -y install openssh-server". The 'purge' command deletes all configuration files and deinstalls the package. I think that openssh may not be properly set up in the image. I tried to move the installation of openssh-server to a later position in the script and have to buy another sd-card to test it.

zerkman, 2016-01-31:
Also got this problem. I confirm parobalth’s solution fixed the issue.

voltagex, 2016-02-12:
@parobalth I'll buy you some SD (MicroSD?) cards or donate if needed.

fangfufu, 2016-02-12:
I can confirm that ''sudo apt-get install --reinstall openssh-server'' fixes the problem.

zerkman, 2016-02-13:
FWIW, I forgot to mention that LXC containers also got terminated when you log out. This is also fixed with the reinstallation of openssh-server.

drtyhlpr, 2016-02-29:
sorry I am very bussy with my job currently . I went to hell with an orchid w board currently working on a g25 project that I know since today should be working start of next week (hey it works very nice so far)...

I will include the fix soon - thanks for your help!

drtyhlpr, 2016-03-01:
hm I currently don't have a rpi2 next to me. but I thought about if the error is related to the following lines (511, 512) of the script:

>   rm -fr $R/etc/ssh/ssh_host_*
>   LANG=C chroot $R dpkg-reconfigure openssh-server

drtyhlpr, 2016-03-12:
checked with latest version of this script -> fixed.thanks for all help

closed 2016-03-12 by drtyhlpr