drtyhlpr/rpi23-gen-image #16: Move initramfs

fangfufu, 2016-02-14:
It appears that the initramfs is located in /boot rather than /boot/firmware. This means that I can't add modules that are needed during the boot process easily - I want to convert my rootfs to btrfs.

drtyhlpr, 2016-03-01:
hi fangfufu, I will try to add an option to cross compile and use a custom linux kernel and optional initramfs in future versions of this script.

fpytloun, 2016-03-10:
I think /boot is correct location for both kernel and initramfs. Package `flash-kernel` is the one responsible for copying kernel into /boot/firmware/kernel7.img and I suppose that it should be done for initramfs as well - so better fix/patch flash-kernel package than hack custom kernel (at least my approach is to be as near to pure Debian as possible).

drtyhlpr, 2016-03-16:
currently working on initramfs fixes

closed 2016-03-16 by drtyhlpr