drtyhlpr/rpi23-gen-image #22: Apt Proxy

stylesuxx, 2016-03-01:
Shouldn't the proxy setting be applied if APT_PROXY is not NULL? **-z**  evaluates to true if the variable is NULL, in this case a string of length 0. In this case I think you should check for not NULL or **-n**

https://github.com/drtyhlpr/rpi2-gen-image/blob/master/rpi2-gen-image.sh#L205

If I set no proxy I get this Error:
```
E: Syntax error /etc/apt/apt.conf.d/10proxy:2: Extra junk at end of file
```

drtyhlpr, 2016-03-03:
fixed by daniell1 / thanks for the fix / I will work on other issues mentioned.

closed 2016-03-03 by drtyhlpr

stylesuxx, 2016-03-03:
Not sure if this solves the problem, since the check with **-z** is still there. This evaluates to true when the string is empty, so the Proxy is only set when the string is empty...

daniell1, 2016-03-03:
My fix only fixed the syntax error. I didn't check anything else :)