drtyhlpr/rpi23-gen-image #23: Calculation of image size

stylesuxx, 2016-03-02:
When calculcating the image file size, something goes wrong.

For example: The build directory is 421612 bytes in size, but I get an image that is 1GB in size.
If I now plug that into the calculation for image size I get the following values
```
CHROOT_SIZE=$(expr 421612 / 1024 ) # = 411
IMAGE_SIZE=`expr $(expr 411 / 1024 + 1) * 1024` # = 1024
IMAGE_SECTORS=`expr $(expr 1024 * 1048576) / 512 - 133120` # = 1964032
```

OK, now this translates to a size limit for the root partition of IMAGE_SIZE - 64 = 960 MB

The steps in which the image "grows" is 1GB, so it is at least of size 1GB.

So this works for image builds where the chroot is below the size of 960MB in total (minus the rootfs).

Now if for example the size of the chroot is 999MB, the resulting image will still be max 1GB in size.
The first 64MB are used for the */boot* partition, leaving 960MB for the rootfs, which in reality is 999MB - (size of the boot partition) which in my case is 22MB. This concludes that I am left with 977MB to write to a partition with 960MB, which results in a "no space left" IO error when copying the files.

My suggestions would be the following:
* Add the 64MB used for */boot* to the initial image size calculation
* Only produce an image of the size that is actually needed
 
I will happily contribute this, if you tell me what you think about it.

drtyhlpr, 2016-03-03:
hi, the image size calc is a bit messy - and I tought about a small "readable" replace. I tought (but it has been I while since I tested it)  - that the image should be "rounded" to 1GB - but the problem with an "edge case" - like 960MB exists. give me some time to test ... I am very bussy these days ... and may take a look this weekend.

drtyhlpr, 2016-03-03:
yeah if you would like to contribute I would be very happy. If I remember right - my idea was to simply add a "min space left free" VARIABLE and simple add this to the calculated "used space" value before the "round" to 1GB

stylesuxx, 2016-03-03:
Hey @drtyhlpr I started playing around with this yesterday. Problem is, that with EXT4 it is very hard to get the *exact* partition size, because of reserved blocks and journaling. So I came up with the following: https://github.com/stylesuxx/rpi2-gen-image/commit/d20b2e032c39241719850e6a96ef28b60419534c I simply add 20% of the chroot size. This should be easily adaptable to a variable with fixed sized instead of percentage. Let me know what you think and I will prepare a PR.

drtyhlpr, 2016-03-04:
I like the patch a lot (math calc on sh is some how a pain in the...) (my alternative idea would be a BC script but this would add another host build dep) but lets simply keep it in sh with you percentage patch and it would be nice if you could create a PR. I want to merge the other(s) PRs soon and do some massive "build testing" (like building images will many different kind of ARGS) - so I can simply 'df' cross check and QS all additional patches and features of the PRs.

stylesuxx, 2016-03-04:
Alright, I'll prepare a PR then. Also my dev branch has fixes for #20 and #21 do you want them too, or should I make seperate PR's for each?

drtyhlpr, 2016-03-04:
please seperate them I am not a merge master and need to learn more about git :)

stylesuxx, 2016-03-04:
Alright, will do that some time over the weekend - my RPi 3 just arrived, and I am a bit distracted ;-)

drtyhlpr, 2016-03-04:
have fun - maybe you wanna try https://github.com/umiddelb/z2d/tree/master/pine64 :)

drtyhlpr, 2016-03-06:
great job / merged / maybe add 25% but auto resize on firstboot should work fine.

closed 2016-03-06 by drtyhlpr