drtyhlpr/rpi23-gen-image #33: Anyone with RPi3? might it be possible to use this script with arch=arm64 (in future)?

drtyhlpr, 2016-03-12:
I do not have a RPi3 yet and I do not know about the status of arm64 rpi3 support. so please tell me your ideas:) ty

fpytloun, 2016-03-13:
It should be possible:
- Debian has arm64 port: https://wiki.debian.org/Arm64Port
- it will be needed to build kernel, probably with `ARCH=arm64` parameter, this is just a small change
- bootloader/firmware will be probably the same?

We just get a new RPi 3 so I can maybe try in a spare time.. :-)

fpytloun, 2016-03-14:
Tried RPiv3 build.
The biggest issue is collabora repo with no arm64 packages. So there's no source to get patched flash-kernel and bootloader packages.

There are more options now:
- contribute to Debian so all required packages will be once upstream (this is correct approach but long-term and requires support from someone who is Debian maintainer)
- build all the required packages and provide them in 3rd party repo (similar to collabora, I already have something at tcpcloud - apt.tcpcloud.eu)
- prepare community-driven build infrastructure and repository for this project (Debian on RPi) but it will then grow from build script to custom distribution

drtyhlpr, 2016-03-14:
thanks for all your ideas and feedback.

https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=138232&p=917951#p91795
hmmm that post sounds bad... . I think patching aarch64 kernel (and videocore) to get it working with RPi3 will take a lot of time and possible jtag hardware debugging...currently for me the only option seems to be to wait to get a working kernel config for `bcm2837_defconfig`..

that one reads better https://www.reddit.com/r/archlinux/comments/48fu0v/help_building_an_arm64_bcm2837raspberry_pi_3_arch/

questions:
- the debootstrap with aarch64 and qemu-aarch64 should work right?le 
- the current RPi3 runs on 32bit mode default and should therefore be compatible to RPi2 "rasbian" and also images created by this script?

about contributing to Debian: I decided for myself that its not possible for me to contribute some official packages or images created by this script. If you want to do this right and responsible - this takes constant time and my life is more like earthquake and fire :)

...argh "Major problem stems from the fact that the VideoCore side of things is still firmly 32-bit."... yeah you mean that closed videocore right? :)

fpytloun, 2016-03-15:
Yes, I also think that 64bit support in kernel + firmware is something what we need to wait for. Anyway bootstraping aarch64 was going fine with my patches linked above.

Original 32bit images should work on RPi3 but didn't - it would probably require newer firmware + kernel to boot. This is what I am currently working on - I have updated packages originally provided by Collabora and Raspbian and made them available here: http://apt.tcpcloud.eu/debian/
So using this repository instead of collabora (to provide latest firmware) and building custom kernel using `BUILD_KERNEL` should work. I will let you know when I succeed :-)

Still I would like to build kernel package the debian way and make it also available in repository.

fpytloun, 2016-03-16:
Ok, so I succesfully built armhf image that is working on RPi3 - using latest kernel and firmware from http://apt.tcpcloud.eu/debian
Not tested very well but it seems that wireless interface is not detected.
And it's still corrupting SD card (Kingston) in the same way as RPi2 with Collabora kernel+firmware.. :-X

drtyhlpr, 2016-03-17:
nice work! I ask myself is these packages should be added to this script to enable some kind of alpha rpi3 support or if it would be possible to compile it "on the fly" like uboot or fbturbo.

ps. (rpi2) I am still struggling with initramfs mmc/sdcard erros. so maybe your sdcard errors are related to that... there is this new driver ... I am doing heavy (sd)debgging the next days because I can not stand it that the script produces and nice bootable system without errors but if I use a simple initramfs I see the sdcard errors...(all of many sandisks that work great anywhere else) hmm I am pretty much sure its not related to modules (missing or wrong options inside initramfs)

pps. ordering rpi3 soon

g-vidal, 2016-06-14:
Hi,
I have got my new raspi3, if you need a dumb beta tester I can take the role :-) . Please let me do what I can do to help.

g-vidal, 2016-07-08:
Hi,
Succeeded install rpi2 imge on my rpi3.... seems to work OK even if there are some clear anomalies.
The reason why there is no WIFI is that raspi has choosen a chipset that needs a proprietary driver  LAN95XX or also called smsc95xx. [microchip](http://www.microchip.com/SWLibraryWeb/producttc.aspx?product=SRC-LAN95xx-LINUX)
The module must be compiled but it failed  because the configuration of my raspi is rather strange...
I would like to try  @fpytloun method with  http://apt.tcpcloud.eu/debian but I don't know how...
I go back to raspbian ;-( but I stay tuned.
If any progress is made I'll check.
thanks again for your work.

closed 2016-08-22 by drtyhlpr