drtyhlpr/rpi23-gen-image #38: Building other debian versions... is that possible ? And issue with language.

g-vidal, 2016-06-13:
Hi,
Many  thanks for this great work, I was wondering if it could be possible to build a debian "stretch" image or more conveniently a "testing" image. I have seen the progress with rpi3 I will get one shortly I can help with tests if you need.

I have also found a bug, the image building process crashes if  I choose a different locale than "en_EN.utf8" I have found a hint using :
`DEFLOCAL="fr_FR.utf8"   ENABLE_MINBASE=true `

I am unfornately not able to propose a clean  patch (I'll try again if I can find some time but I am not a dev...)

drtyhlpr, 2016-06-13:
Hi, thanks for you feedback! I am currently busy with other stuff and will try to debug your identified problem possibly next weekend! 

To be honest I forgot to order and take care of rpi3, I am not even aware of the official 64-bit support "status" :.-) I would love to see this script running on other arm or arm64 boards (beaglebone, banana etc) - some have a kind of weak up-to-date kernel support ...hm - but I am also aware that this will exp enhance the amount of time needed to take care (test, patch etc) of the script.

drtyhlpr, 2016-06-13:
sorry but I think I was to tiered to see the problem :-)

cause of the problem: DEFLOCAL does not work in combination with ENABLE_MINBASE!

README: "Set default system locale. This setting can also be changed inside the running OS using the dpkg-reconfigure locales command. The script variant minbase (ENABLE_MINBASE=true) doesn't install locales."

g-vidal, 2016-06-14:
Yes that's the point !
The trick I used was to deactivate the DEFLOCAL option, I had to reconfigure console keyboard and locales ;-(
My problem is that with
`DEFLOCAL="fr_FR.utf8"`
The scrip crashes and does not produce the image
I'll start again and provide the error message ASAP

drtyhlpr, 2016-06-18:
I will simply try to create a small check. In my opinion if you want to create a minimal image (using MINBASE "only" reduces the regular image size by 64MB) fe. if you use X and or any kind of web browser on the system do NOT use MINBASE because you will not safe 1MB of space - because 96% of the skipped packages of MINBASE will get installed as dependencies.

local(s) is/are huge and not compatible with the idea of a MINBASE (minimized) image.

drtyhlpr, 2016-06-19:
hi, this should work now with the latest commit. please test.

ps.I realized during testing: make sure to use the right LANG definition: fr_FR.UTF**-**8

g-vidal, 2016-06-19:
Sorry for the mistake on language code !
evrything works properly now  I'll close the issue :-)

I am still looking  forward to be able to build  a  stretch image and a stretch image on a pi3 but it is another story ....

Sorry again and thanks again for your time

closed 2016-06-19 by g-vidal

drtyhlpr, 2016-06-19:
no problem - thanks for the feedback so I could enhance the code