drtyhlpr/rpi23-gen-image #39: No i2c dev available with the constructed image (device-tree problem)

g-vidal, 2016-06-14:
Hi 
I cannot set up /dev/i2c* devices in the raspberry with the image built with rpi2-gen-image... I have tried allt ricks I have found on the web  with no effect and finally I asked the question on Rasberry forum and the engineer sends me back here with the following message : [Raspberry Pi forum](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=97314&start=250) 

> The dtparam directive in config.txt constructs a device tree in memory, then passes its address to the kernel. If your kernel is u-boot then it is likely that the message isn't getting through. You'll have to ask the maintainers of your distribution what to do. If your u-boot is expecting the device-tree at 0x100 then you might be able to use:

    `device_tree_address=0x100`

in config.txt. The correct technique is for u-boot to either specify the required DTB load address using a custom stub file or to read the actual load address from address 0xf8 (see [raspberrypi/tools](https://github.com/raspberrypi/tools/tree/master/armstubs)    )

> Device Tree is required when using our 4.4 kernels, so attempts to disable it will be ignored.

This comment is beyond my competencies, but I suppose that some other users may like to be able to use i2c bus on their raspberry , I hope this comment may help to solve the problem...

Thanks again for your work

drtyhlpr, 2016-06-18:
thanks for your error report, I will take a look the following days. I agree i2c is essential for most oft the users! this device tree stuff is somehow strange to me sometimes (I would enable device tree debugging stuff etc). I remember that I have tried to fully disable sound using device tree, and I gave up after a couple of days .... (current state: not possible)

drtyhlpr, 2016-06-18:
quick fix idea (not able to verify currently):

```
device_tree_address=0x100
device_tree_end=0x8000
```

ps. I have read parts of these forums posts.. and its just a big mess - like systemd...if BSD would have a better support I would have create a bootsrapping script for BSD. sorry for the sarcasm but this device tree (integration) is real a mess...

g-vidal, 2016-06-19:
The image has been built successfully
During the process I have seen the line
 LD      drivers/i2c/built-in.o
french locale has been built (sorry for the mistake  on the language code)
and is perfectly operational at first boot :-)

I have got my i2c bus with the following configurations : 
```
# boot u-boot kernel
#kernel=u-boot.bin
device_tree_address=0x100
device_tree_end=0x8000
dtparam=i2c_arm=on
```
I have not yet tested if it operates properly but now it is present which solves the issue.
It seems that it is not possible to get it with u-boot on..... but that's a secondary issue :-)

many thanks for the job!

closed 2016-06-19 by g-vidal

drtyhlpr, 2016-06-19:
ok great - i will test the i2c stuff this evening and add it to the default configuration and or testing uboot config. thank you aggain