drtyhlpr/rpi23-gen-image #40: Upcoming Debian stretch support

drtyhlpr, 2016-06-20:
I did first tests to support the Debian "stretch" release. You can try to create a bootable (but unstable) system using RELEASE=stretch parameter. Currently the init, systemd and systemd-sysv packages cause trouble with the "stretch" release.

I used `RELEASE=stretch ENABLE_MINBASE=true APT_INCLUDES="systemd"`
 and added a simple apt-get install systemd-sysv at the end of the script.

So far no success on building (default) ENABLE_MINBASE=false "stretch" releases. Feel free to support.

xtaran, 2016-06-20:
What about using `APT_INCLUDES="init"` instead of mentioning `systemd` and `systemd-sysv` explicitly? `init` depends (with alternatives) on `systemd-sysv` and that depends on `systemd`. (`init` recently has been downgraded in Sid/Stretch from `required` plus `essential` to `important` because it's not needed in chroot and some container environements. Maybe that's causing these issues.)

drtyhlpr, 2016-06-20:
Hi - Sorry I don't know. I currently don't have time to test. I see couple of errors (apt-key stuff) and errors with systemd-sysv on MINBASE (couple more on not MINBASE) and no useful info from bootstrap.log (nice systemd-sysv depends on systemd loop). Installing systemd-sysv outside (afterwards) the APT_INCLUDES/debootstrap works but I did not find time to investigate why.

I am a little bit frustrated about the latest serial console problems (happening on stretch AND jessie), hard to debug ... not very motivating. and a great time killer. pritty sure related to systemd (yeah yeah I ll "read the changelog"(tm) - but that systemd integration more seems like a assimilation - do it our new way or die)

drtyhlpr, 2016-06-20:
also with using APT_INCLUDES="init" the problem seems to be with "systemd-sysv". this creates an error on debootstrap based installation "systemd installed but not configured" - and keeps looping a couple of times.

I an not sure whats wrong but the bootstrap package installation seems "to need systemd to be configured". And I think it is configured at a later (final step) of the bootstrap...

drtyhlpr, 2016-08-06:
stretch release support added.

closed 2016-08-06 by drtyhlpr