drtyhlpr/rpi23-gen-image #44: Bad depends for Stretch

mati75, 2016-08-07:
Hi,

On build RELEASE=stretch ENABLE_MINBASE=true ENABLE_REDUCE=true ENABLE_MINGPU=true BUILD_KERNEL=true ENABLE_UBOOT=true I have:

`+ chroot_exec apt-get -q -y --force-yes --no-install-recommends install linux-compiler-gcc-4.8-arm g++ make bc
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /home/mati75/rpi2-gen-image/images/stretch/build/chroot apt-get -q -y --force-yes --no-install-recommends install linux-compiler-gcc-4.8-arm g++ make bc
Reading package lists...
Building dependency tree...
E: Unable to locate package linux-compiler-gcc-4.8-arm
E: Couldn't find any package by glob 'linux-compiler-gcc-4.8-arm'
E: Couldn't find any package by regex 'linux-compiler-gcc-4.8-arm'
`
Package in stretch have name linux-compiler-gcc-5-arm.

drtyhlpr, 2016-08-07:
oh thanks for you feedback - I will check it, so the UBOOT dep gcc is wrong and propaply just needs a new version name.

drtyhlpr, 2016-08-07:
yes - I need to patch the apt-get package names in 41-uboot amd 42-fbturbo for stretch release. should be fixed soon.

drtyhlpr, 2016-08-08:
errors on ENABLE_UBOOT, ENABLE_FBTURBO, ENABLE_REDUCE fixed. have fun

closed 2016-08-08 by drtyhlpr