drtyhlpr/rpi23-gen-image #45: Scripts work for raspi2 & raspi3 -> request for enhancements (please...)

g-vidal, 2016-08-08:
Hi all,
This script works like a charm! The image produced can run identically on _Raspi2_ **AND** _Raspi3_.
For display manager and window manager I have tested lightdm and lxde (it works perfectly, tested in its minimalist form without background and lxde "goodies").
Thanks for enabling the non-free which permits to use much easily the wifi but some improvements will reduce the necessary initial manipulations (see below).
I used the following parameters to build my image among other more specific (name,login country, etc...): 
```shell
APT_INCLUDES="firmware-linux-free,firmware-linux-nonfree,firmware-brcm80211,firmware-realtek,network-manager,network-manager-gnome,lxde" 
ENABLE_NONFREE=true 
ENABLE_WM="lightdm" 
```
Things that may help a lot : 
- When the non-free firmware packages are added, the raspi starts the wifi but unfortunately it cannot connect to it because it needs some config parameters that can be given when it has booted with the wire... It should be very interesting if such parameters could be passed through the script because in that case no need of a wired connection any more to start! For exemple it might be possible for network-manager users to provide a file from `/etc/NetworkManager/system-connections`  or its contents 
- Unfortunately the driver of the raspi3 wifi is not available and the only way to get it I have found is to copy it from raspbian :-(  see [Raspberry forum](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=141834)

> 
I think you'll need to lift /lib/firmware/brcm/* from a Raspbian card.
I've put a 7zipped version on Github at: [DougieLawson](https://github.com/DougieLawson/RaspberryPi/tree/master/brcm)

I do not know what will be the proper way to handle this but it is frustrating to se that the correct driver is there but it does not work and finally the only solution is to copy Raspbian's one ! (I spent too much time finding that even if the driver name was correct the content was not! :-( )

Thanks for this very usefull work, hope you may find some time to consider my proposals.

Best regards.

drtyhlpr, 2016-08-08:
Hi,

thanks for you feedback! hm I will think about how to implement "static ip parameters" for wifi cards or other custom network adapters- maybe some "input systemd network conf" that the user can specify. might take some time. I dont know about the rpi3 wifi drivers but I will update my brain on that, maybe I can help then.

I am happy with your proposals and other feedbacks!

have a nice day!

drtyhlpr, 2016-08-09:
I am in holidays and don't have a rpi3 near, but ill test it soon. I am a alittle bit confused the same kernel runs on both rpi2 and rpi3 because the Kernel compilation currently uses the following options:

Device Tree: rpi-2-b.dtb
Kernel def conf: bcm2709_defconfig

https://github.com/raspberrypi/linux/tree/rpi-4.4.y/arch/arm/configs
https://github.com/raspberrypi/linux/tree/rpi-4.4.y/arch/arm/boot/dts

I still can find any RPI3 / BCM2837 specific defconf or dtb. hmmm

drtyhlpr, 2016-08-09:
just got my hands on rpi3 - working on pull request. released soon

drtyhlpr, 2016-08-10:
currently doing lasts tests with the automatic wireless driver download. rpi3 support is 95% done - but I have problems booting with uboot bootloader, rpi default bootloader works fine.

g-vidal, 2016-08-22:
Hi,  Thanks a lot !!

I have tested your new script on both pi2 and pi3 under **stretch** It works perfectly. I can confirm that it is possible to buid an image using lxdm and lxde in their minimalistic form. The WIFI is working very well on both types.

 I have written a  [kind of tutorial in french](http://blog.climatetmeteo.fr/GerardVidal/raspi23Stretch_install.html) that cites and relies on your work tell me if you are OK with it. The aim is to produce a rpi2 and rpi3 stretch image for a weatherstation project that enables weewx mraa and upm utilities and libraries on the raspberry to handle sensors on the GPIO.

_Remarks and suggestions_
It should be usefull to put the default user in the netdev group even for rpi2 (when the wifi cannot be enabled by default) because when the WIFI is installed one may not spontaneously think that the initial (privileged) user is not member of that group.

I had a dependency problem with gnupg1 / gnupg. I am not sure if it has been introduced by one of he packages I have added or by a version change in gnupg or gnupg1.
I solved the issue by adding gnupg and gnupg2 in APT_INCLUDES.

Thanks again

drtyhlpr, 2017-01-10:
@g-vidal I am closing this "issue" now. If you have further ideas or feature request just tell me :)

closed 2017-01-10 by drtyhlpr