drtyhlpr/rpi23-gen-image #46: Is it possible to add during the build a package that is not in the debian repositories ?

g-vidal, 2016-08-08:
Hi,
I am using the Raspberry to share meteorological data from a weatherstation. I am using weewx software to get data from the WeatherStation and unfortunately, even if it is very active and opensource, the software is not in the debian repositories. A .deb package can be dowloaded from [weewx site](http://weewx.com/downloads/).
Would it be possible to include this deb file in the process of building the image to have it installed automatically (Maybe also usefull for other makers that may have "non-standard packages").

Thanks again, best regards

drtyhlpr, 2016-08-08:
Hi,
"Debian custom packages, i.e. those not in the debian repositories, can be installed by placing them in the 'packages' directory. They are installed immediately after packages from the repositories are installed."

https://github.com/drtyhlpr/rpi2-gen-image/commit/d4a066f30735ea721a85f210f9a7526bee144fd9

the following commit shows (the few) relevant lines. so in general (for every deb in packages directory):
dpkg --unpack .deb
apt-get -qq -y -f install

to avoid errors you can first verify if these commands work fine on a fresh running rpi2 image.

drtyhlpr, 2016-08-08:
just tested and seems to work
mkdir packages
cd packages
wget http://weewx.com/downloads/weewx_3.5.0-1_all.deb
cd ..

run script ->
Starting weewx weather system: weewx.

g-vidal, 2016-08-09:
Many thanks 
I'll double check and modify my image creation.

closed 2016-08-10 by drtyhlpr