drtyhlpr/rpi23-gen-image #48: Wrong links in /lib/modules/$(uname -r)

simark, 2016-09-07:
The `build` and `source` links in `/lib/modules/$(uname -r)` have a destination that is on the build machine:

```
ls -l build source 
lrwxrwxrwx 1 root root 82 Sep  7 16:07 build -> /home/simark/polarsys-os/rpi23-gen-image/images/stretch/build/chroot/usr/src/linux
lrwxrwxrwx 1 root root 82 Sep  7 16:07 source -> /home/simark/polarsys-os/rpi23-gen-image/images/stretch/build/chroot/usr/src/linux
```

As a result, it's not possible to build external kernel modules.  The workaround is simply to make the links point at the right place.

In my setup, I use a kernel that I build myself beforehand (`KERNELSRC_PREBUILT=true`). I guess these links are installed by the `make modules_install` step?  Perhaps it needs to be done from inside the chroot, or the links need to be fixed up afterwards?

drtyhlpr, 2016-09-10:
hi thanks for your feedback, i will investigate the problem the next days.  can you tell me what external module u are including - so I can use this for my testing.

simark, 2016-09-10:
Hi drtyhlpr,

The modules in question are lttng-modules: https://github.com/lttng/lttng-modules
Once the kernel headers/sources are properly installed, you just need to type `make` to build them (and `make modules_install` to install them in /lib/modules).

Note that in the mean time I found the option `KERNEL_REMOVESRC=false`, which leaves the kernel sources in `/usr/src/linux`.  So in order to build the modules, I first needed to adjust the `/lib/modules/$(uname -r)/build` and `/lib/modules/$(uname -r)/source` links to point to `/usr/src/linux`.

However, i stumbled on another problem, which is that some tools in `/usr/src/linux/scripts` were built for x86-64, since the kernel was compiled on an x86-64 machine.  Deleting those scripts and running `make scripts` and/or `make prepare` in `/usr/src/linux` rebuilt them for ARM.  I was then able to build the lttng modules properly.

So in order to be able to build external modules out of the box on the target, I only see those two issues:
* Links in `/lib/modules/$(uname -r)` point to nowhere.
* Tools in `/usr/src/linux/scripts` built for x86-64.

It would also be nice to be able to build and install external modules when building the image.  I think it would be fairly easy to cross-compile them (at least when we know where the kernel sources are) just like we cross-compile the kernel.

Thanks for this script, it's really useful!

drtyhlpr, 2016-09-13:
Hi simark,

thanks for all the detailed information I will work on a fix soon - I have a ugly deadline for other stuff till end of next week -  I will focus on the patch afterwards! And of course thanks for your feedback it helps to make the script better and better!

ps. usefull: try the full encrypted rootfs with "amazing" r/w speed, a quite unique feature for a bootstrapping script with image file output - never seen this before so far and it was a big pain to get it running with auto-resize :)

drtyhlpr, 2017-01-05:
I think this should work now :) https://github.com/drtyhlpr/rpi23-gen-image/commit/a38e66ced2d0ffdc6fccd00bb81ba6fba6fda13c

closed 2017-01-05 by drtyhlpr