drtyhlpr/rpi23-gen-image #49: Config to activate I2C on Raspberry PI3

g-vidal, 2016-09-11:
Hi all,
In the image built with rpi23-gen-image I2C bus is not activated. To do so things have slightly changed from rpi2 situation. Two things must be done :

1. enable the i2c modules by un commenting 2 lines in the `/lib/modules-load.d/rpi2.conf` file 
```shell
# bcm2708_rng
snd_bcm2835
i2c-bcm2708
i2c-dev
# rtc-ds1307
```
2. add the following lines to `/boot/firmware/config.txt` file
```shell
dtoverlay=i2c0-bcm2708
dtoverlay=i2c1-bcm2708
```
Then reboot and you'll get the wanted `/dev/i2c-0` and `/dev/i2c-1` devices; of course if you need just one bus activate only one line.

```
sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- 18 -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- 77                         

```

Best regards

drtyhlpr, 2016-09-13:
Hi g-vidal,

thanks again for your feedback. I will probably add a ENABLE_I2C argument to the script soon. currently bussy with other stuff and sadly I don't have a Debian system around currently.

have a nice day

drtyhlpr, 2017-01-23:
hi @g-vidal !

I just added ENABLE_I2C and ENABLE_SPI options. Can you please test it? (I will not have a i2c or spi dev available until next weekend).

https://github.com/drtyhlpr/rpi23-gen-image/commit/fa587c0f5aa0015e19d351f69e07e4ea4c1c84f2

drtyhlpr, 2017-01-28:
I tested I2c seems to work. have fun

closed 2017-01-28 by drtyhlpr