drtyhlpr/rpi23-gen-image #50: [RPI_MODEL=2] It doesn't boot - 7 green blink

28446, 2016-09-13:
[build-22431209.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/469255/build-22431209.txt)
Hi,
I've built the image yesterday evening whit the attached parameters, when I plug the power chord the rasp doesn't boot.
According whit this table[1], rasp is saying "kernel.img not found", but looking in the /boot mntpoint i see these files:

-rw-r--r-- 1 root root   17840 dic 28  2014 bootcode.bin
-rw-r--r-- 1 root root     192 set 12 23:01 cmdline.txt
-rw-r--r-- 1 root root    1311 set 12 23:01 config.txt
-rw-r--r-- 1 root root    2352 dic 28  2014 fixup_cd.dat
-rw-r--r-- 1 root root    6140 dic 28  2014 fixup.dat
-rw-r--r-- 1 root root    9188 dic 28  2014 fixup_x.dat
-rw-r--r-- 1 root root 3304216 set 12 23:01 kernel7.img
-rw-r--r-- 1 root root  550584 dic 28  2014 start_cd.elf
-rw-r--r-- 1 root root 2636696 dic 28  2014 start.elf
-rw-r--r-- 1 root root 3593672 dic 28  2014 start_x.elf

Following some information I hope can help you:

The host where I've built the image is a debian stretch:

debootstrap:all/testing 1.0.81 uptodate
debian-archive-keyring:all/testing 2014.3 uptodate
qemu-user-static:amd64/testing 1:2.6+dfsg-3 uptodate
binfmt-support:amd64/testing 2.1.6-1 uptodate
dosfstools:amd64/testing 4.0-2 uptodate
rsync:amd64/testing 3.1.1-3 uptodate
bmap-tools:all/testing 3.2-4 uptodate
whois:amd64/testing 5.2.12 uptodate
git:amd64/testing 1:2.9.3-1 uptodate

The card is a Raspberry Pi 2 B+ V 1.2 (2014)

My parameters:

script -c 'RPI_MODEL=2 ENABLE_CONSOLE=true ENABLE_DHCP=true APT_INCLUDES="firmware-linux-free,network-manager" TIMEZONE="Europe/Rome" XKB_LAYOUT="it" USER_NAME=pi ./rpi23-gen-image.sh' 
./build-$(date +%H%M%d%m)

This seems not help, 7 green blink:

APT_INCLUDES="firmware-linux-free,firmware-linux-nonfree,firmware-brcm80211,firmware-realtek,network-manager" 
ENABLE_NONFREE=true 

This one not work, when I enable UBOOT rasp blink one green and then nothing:

ENABLE_UBOOT=true

To try out the SD and rasp integrity, after the rpi23-gen-image test I've flashed 2016-05-27-raspbian-jessie-lite.img in the same SD and it work.

[1] http://elinux.org/R-Pi_Troubleshooting#Green_LED_blinks_in_a_specific_pattern

drtyhlpr, 2016-09-13:
Hi !

thanks for your detailed information and build log. I will not be able to patch it before end of next week. Can you (maybe as work around) try to build the latest Kernel via BUILD_KERNEL=true? The default static, precompiled kernel is very outdated: https://repositories.collabora.co.uk/debian/pool/rpi2/l/linux/.

28446, 2016-09-13:
> Can you (maybe as work around) try to build the latest Kernel via BUILD_KERNEL=true? The default static, precompiled kernel is very outdated: https://repositories.collabora.co.uk/debian/pool/rpi2/l/linux/.

Sure, I'm testing just now whit:

`script -c 'RPI_MODEL=2 RELEASE="jessie" ENABLE_CONSOLE=true ENABLE_DHCP=true BUILD_KERNEL=true KERNEL_THREADS=3 KERNELSRC_DIR="/home/felipe/rpi23-gen-image/linux" APT_INCLUDES="firmware-linux-free,network-manager" TIMEZONE="Europe/Rome" XKB_LAYOUT="it" USER_NAME=dipa ./rpi23-gen-image.sh' ./build-$(date +%H%M%d%m)`

I've previously took the source from git, with `git clone https://github.com/raspberrypi/linux`
It still doesn't boot, green led blink 7 times.

Following the content of /boot in 2016-05-27-raspbian-jessie-lite:

ll /home/felipe/2016-05-27-raspbian-jessie-lite/boot
totale 20312
-rwxr-xr-x 1 root root   13065 mag 27 12:20 bcm2708-rpi-b.dtb
-rwxr-xr-x 1 root root   13328 mag 27 12:20 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 root root   13101 mag 27 12:20 bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 root root   14411 mag 27 12:20 bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root   15081 mag 27 12:20 bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root   17932 mag 27 12:20 bootcode.bin
-rwxr-xr-x 1 root root     184 mag 27 14:15 cmdline.txt
-rwxr-xr-x 1 root root    1635 mag 27 14:09 config.txt
-rwxr-xr-x 1 root root   18693 ago 21  2015 COPYING.linux
-rwxr-xr-x 1 root root    2509 mag 27 12:20 fixup_cd.dat
-rwxr-xr-x 1 root root    6480 mag 27 12:20 fixup.dat
-rwxr-xr-x 1 root root    9713 mag 27 12:20 fixup_db.dat
-rwxr-xr-x 1 root root    9713 mag 27 12:20 fixup_x.dat
-rwxr-xr-x 1 root root     145 mag 27 14:41 issue.txt
**-rwxr-xr-x 1 root root 4224096 mag 27 12:20 kernel7.img
-rwxr-xr-x 1 root root 4122560 mag 27 12:20 kernel.img**
-rwxr-xr-x 1 root root    1494 nov 18  2015 LICENCE.broadcom
-rwxr-xr-x 1 root root   18974 mag 27 14:41 LICENSE.oracle
drwxr-xr-x 2 root root    8192 mag 27 14:08 overlays
-rwxr-xr-x 1 root root  615896 mag 27 12:20 start_cd.elf
-rwxr-xr-x 1 root root 4934920 mag 27 12:20 start_db.elf
-rwxr-xr-x 1 root root 2743224 mag 27 12:20 start.elf
-rwxr-xr-x 1 root root 3887400 mag 27 12:20 start_x.elf

compared whit the content of /boot in 2016-09-13-debian-jessie.img:

ll
totale 11338
-rw-r--r-- 1 felipe felipe   15127 set 13 14:05 bcm2709-rpi-2-b.dtb
-rw-r--r-- 1 felipe felipe   15763 set 13 14:05 bcm2710-rpi-3-b.dtb
-rw-r--r-- 1 felipe felipe   15114 set 13 14:05 bcm2710-rpi-cm3.dtb
-rw-r--r-- 1 felipe felipe   17932 set 13 14:05 bootcode.bin
-rw-r--r-- 1 felipe felipe     192 set 13 14:05 cmdline.txt
-rw-r--r-- 1 felipe felipe    1311 set 13 14:05 config.txt
-rw-r--r-- 1 felipe felipe    2522 set 13 14:05 fixup_cd.dat
-rw-r--r-- 1 felipe felipe    6498 set 13 14:05 fixup.dat
-rw-r--r-- 1 felipe felipe    9732 set 13 14:05 fixup_x.dat
**-rw-r--r-- 1 felipe felipe 4212444 set 13 14:05 kernel7.img**
drwxr-xr-x 2 felipe felipe    8192 set 13 14:05 overlays
-rw-r--r-- 1 felipe felipe  632984 set 13 14:05 start_cd.elf
-rw-r--r-- 1 felipe felipe 2763160 set 13 14:05 start.elf
-rw-r--r-- 1 felipe felipe 3894648 set 13 14:05 start_x.elf

I'm wondering why there are two kernel img in raspbian /boot partition and if this it could be 
the problem (I'm sorry form my English).

[build-13311309.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/469776/build-13311309.txt)

drtyhlpr, 2016-09-13:
hm strange... 

in lines bootstrap.d/13-kernel.sh:123 and 126 you can see that the compiled kernel zImage get copied (or converted an the copied) to ${KERNEL_IMAGE} (that is kernel7.img). 

on the page https://pi-ltsp.net/advanced/kernels.html it states that

> The kernel for the Raspberry Pi can be found on the /boot partition (the FAT32 partition you can open on Windows, Mac and Linux).
> For the Raspberry Pi 1 (B, B+, A, A+) it is called kernel.img.
> For the Raspberry Pi 2 (B) it is called kernel7.img.

so this would explain why there is also a kernel.img on the rasbian-jessie image. but I am a little bit out of ideas about your problem. (so ill simply need time with hand on a debian host and rpi2).

hm last "pronto" idea I have - the error also appears if you leave out all your custom parameters ? 
so the minimal parameters create the same error ? (I am just trying to sort out possible error sources :)

`
BUILD_KERNEL=true KERNELSRC_DIR=yourdir ./rpi23-gen-image.sh
`

28446, 2016-09-14:
> hm last "pronto" idea I have - the error also appears if you leave out all your custom parameters ?
so the minimal parameters create the same error ? (I am just trying to sort out possible error sources :)

BUILD_KERNEL=true KERNELSRC_DIR=yourdir ./rpi23-gen-image.sh

To keep it simple I've built the image with:

`script -c 'BUILD_KERNEL=true ./rpi23-gen-image.sh' ./build-$(date +%H%M%d%m)`

It doesn't boot, 7 green blink

Thank you
  
[build-09391409.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/471594/build-09391409.txt)

28446, 2016-09-20:
> ps. "The host where I've built the image is a debian stretch", hm I hope / am sure - this ll not be a problem but to be honest I only tested this script on jessie and older so far.

Among various tests I have built the image on a jessie vm and I found that the script dependencies lead to install crossbuild-essential-armhf that is in testing/unstable, not in jessie. However, it 
doesn't help ;/

This is the structure of raspbian img:

Disk 2016-05-27-raspbian-jessie-lite/2016-05-27-raspbian-jessie-lite.img: 1,3 GiB, 1387266048 bytes, 2709504 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x6fcf21f3

Device                                                               Boot  Start     End Sectors  Size Id Type
2016-05-27-raspbian-jessie-lite/2016-05-27-raspbian-jessie-lite.img1        8192  137215  129024   63M  c W95 FAT32 (LBA)
2016-05-27-raspbian-jessie-lite/2016-05-27-raspbian-jessie-lite.img2      137216 2709503 2572288  1,2G 83 Linux

I've tried to conform the structure of debian jessie img to the raspibian img, toggling to off boot flag and shifting forward to 8192 TABLE_SECTORS, but it doesn't help. 

Disk 2016-09-20-debian-jessie.img: 516 MiB, 541022208 bytes, 1056684 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device                                  Boot  Start     End Sectors  Size Id Type
2016-09-20-debian-jessie.img1 *      2048  133119  131072   64M  c W95 FAT32 (LBA)
2016-09-20-debian-jessie.img2      133120 1056683  923564  451M 83 Linux

When I run fsck it say it's all ok

I've flashed raspian yesterday to prove the SD integrity, it (still) works fine.

I'm confused, i guess it is a kernel problem but I' don't know how to help you.

When i set `kernel=kernel7.img` in `config.txt` the green LED blip once and nothing more, the page I've previously mentioned reports:

`Older images do not load boot code for revB boards with the Hynix chip. Use release 2013-02-09 (?) or later. (I observe a single blip on the green activity LED)`

I attach to this comment some info I've collected whit raspbian, I hope that can be useful.

[config.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/483132/config.txt)
[cmdline.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/483131/cmdline.txt)
[cpuinfo.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/483136/cpuinfo.txt)
[bootstrap.log.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/483137/bootstrap.log.txt)
[lsusb.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/483139/lsusb.txt)
[dmesg.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/483140/dmesg.txt)
[kern.log.txt](https://github.com/drtyhlpr/rpi23-gen-image/files/483141/kern.log.txt)

28446, 2016-09-20:
I'm so sorry, I'm wrong, mine is a Raspberry Pi 1.

drtyhlpr, 2017-01-05:
argh!

closed 2017-01-05 by drtyhlpr