drtyhlpr/rpi23-gen-image #56: Unable to locate package crossbuild-essential-armhf

Alexrocksdude, 2016-10-22:
Not sure what else I can add other than the output:

``alex@debian:~/rpi23-gen-image$    echo y | sudo RPI_MODEL=3 ./rpi23-gen-image.sh
-e 
#
#RPi2/3 Bootstrap Settings
#
+ RPI_MODEL=3
+ RPI2_DTB_FILE=bcm2709-rpi-2-b.dtb
+ RPI2_UBOOT_CONFIG=rpi_2_defconfig
+ RPI3_DTB_FILE=bcm2710-rpi-3-b.dtb
+ RPI3_UBOOT_CONFIG=rpi_3_32b_defconfig
+ RELEASE=jessie
+ KERNEL_ARCH=arm
+ RELEASE_ARCH=armhf
+ CROSS_COMPILE=arm-linux-gnueabihf-
+ COLLABORA_KERNEL=3.18.0-trunk-rpi2
+ KERNEL_DEFCONFIG=bcm2709_defconfig
+ KERNEL_IMAGE=kernel7.img
+ QEMU_BINARY=/usr/bin/qemu-arm-static
+ KERNEL_URL=https://github.com/raspberrypi/linux
+ FIRMWARE_URL=https://github.com/raspberrypi/firmware/raw/master/boot
+ WLAN_FIRMWARE_URL=https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm80211/brcm
+ COLLABORA_URL=https://repositories.collabora.co.uk/debian
+ FBTURBO_URL=https://github.com/ssvb/xf86-video-fbturbo.git
+ UBOOT_URL=git://git.denx.de/u-boot.git
+ pwd
+ BASEDIR=/home/alex/rpi23-gen-image/images/jessie
+ BUILDDIR=/home/alex/rpi23-gen-image/images/jessie/build
+ R=/home/alex/rpi23-gen-image/images/jessie/build/chroot
+ ETC_DIR=/home/alex/rpi23-gen-image/images/jessie/build/chroot/etc
+ LIB_DIR=/home/alex/rpi23-gen-image/images/jessie/build/chroot/lib
+ BOOT_DIR=/home/alex/rpi23-gen-image/images/jessie/build/chroot/boot/firmware
+ KERNEL_DIR=/home/alex/rpi23-gen-image/images/jessie/build/chroot/usr/src/linux
+ WLAN_FIRMWARE_DIR=/home/alex/rpi23-gen-image/images/jessie/build/chroot/lib/firmware/brcm
+ RPI_FIRMWARE_DIR=
+ HOSTNAME=rpi3-jessie
+ PASSWORD=raspberry
+ DEFLOCAL=en_US.UTF-8
+ TIMEZONE=Europe/Berlin
+ EXPANDROOT=true
+ XKB_MODEL=
+ XKB_LAYOUT=
+ XKB_VARIANT=
+ XKB_OPTIONS=
+ ENABLE_DHCP=true
+ NET_ADDRESS=
+ NET_GATEWAY=
+ NET_DNS_1=
+ NET_DNS_2=
+ NET_DNS_DOMAINS=
+ NET_NTP_1=
+ NET_NTP_2=
+ APT_PROXY=
+ APT_SERVER=ftp.debian.org
+ ENABLE_CONSOLE=true
+ ENABLE_IPV6=true
+ ENABLE_SSHD=true
+ ENABLE_NONFREE=false
+ ENABLE_WIRELESS=false
+ ENABLE_SOUND=true
+ ENABLE_DBUS=true
+ ENABLE_HWRANDOM=true
+ ENABLE_MINGPU=false
+ ENABLE_XORG=false
+ ENABLE_WM=
+ ENABLE_RSYSLOG=true
+ ENABLE_USER=true
+ USER_NAME=pi
+ ENABLE_ROOT=false
+ ENABLE_ROOT_SSH=false
+ ENABLE_MINBASE=false
+ ENABLE_REDUCE=false
+ ENABLE_UBOOT=false
+ ENABLE_FBTURBO=false
+ ENABLE_HARDNET=false
+ ENABLE_IPTABLES=false
+ ENABLE_SPLITFS=false
+ ENABLE_INITRAMFS=false
+ ENABLE_IFNAMES=true
+ BUILD_KERNEL=false
+ KERNEL_REDUCE=false
+ KERNEL_THREADS=1
+ KERNEL_HEADERS=true
+ KERNEL_MENUCONFIG=false
+ KERNEL_REMOVESRC=true
+ KERNELSRC_DIR=
+ KERNELSRC_CLEAN=false
+ KERNELSRC_CONFIG=true
+ KERNELSRC_PREBUILT=false
+ REDUCE_APT=true
+ REDUCE_DOC=true
+ REDUCE_MAN=true
+ REDUCE_VIM=false
+ REDUCE_BASH=false
+ REDUCE_HWDB=true
+ REDUCE_SSHD=true
+ REDUCE_LOCALE=true
+ ENABLE_CRYPTFS=false
+ CRYPTFS_PASSWORD=
+ CRYPTFS_MAPPING=secure
+ CRYPTFS_CIPHER=aes-xts-plain64:sha512
+ CRYPTFS_XTSKEYSIZE=512
+ DISABLE_FBI=false
+ CHROOT_SCRIPTS=
+ APT_INCLUDES=
+ APT_INCLUDES=,apt-transport-https,apt-utils,ca-certificates,debian-archive-keyring,dialog,sudo,systemd,sysvinit-utils
+ REQUIRED_PACKAGES=debootstrap debian-archive-keyring qemu-user-static binfmt-support dosfstools rsync bmap-tools whois git
+ MISSING_PACKAGES=
+ set +x
dpkg-query: no packages found matching crossbuild-essential-armhf
the following packages needed by this script are not installed:
 crossbuild-essential-armhf

do you want to install the missing packages right now? [y/n] E: Unable to locate package crossbuild-essential-armhf``

shaardie, 2016-11-04:
I have installed it under Debian Jessie using the following: https://wiki.debian.org/CrossToolchains#Installation

It also seems to be able to install it native in Debian Stretch https://packages.debian.org/stretch/crossbuild-essential-armhf

Maybe it can help...

drtyhlpr, 2017-01-05:
hm sorry for the delay. if its still important: what host linux os are you using?

https://wiki.debian.org/CrossToolchains#For_jessie_.28Debian_8.29

denismosolov, 2017-01-05:
I got a similar issue on Ubuntu 16.04. I replaced crossbuild-essential-armhf with gcc-arm-linux-gnueabihf and it's gone:

https://github.com/denismosolov/rpi2-gen-image/commit/f885521a0f43845b9c14e1424b136c8400f2123e

drtyhlpr, 2017-01-05:
oh I wasn't even aware that this script able to run on a ubuntu build host hehe. I am just working on a patch to get it running on a "fresh installed" jessie and stretch with "standard system utilities" packages.

so if you @Alexrocksdude  are using a Debian jessie build host you need to do the following:
```
echo "deb http://emdebian.org/tools/debian/ jessie main" > /etc/apt/sources.list.d/crosstools.list
sudo -u nobody wget -O - http://emdebian.org/tools/debian/emdebian-toolchain-archive.key | apt-key add -
dpkg --add-architecture armhf
apt-get update
```

quozl, 2017-01-06:
i am reviewing prior to using rpi23-gen-image.  i was about to try ubuntu, but will focus on debian first.  i couldn't find build host specs in the readme.

denismosolov, 2017-01-06:
@drtyhlpr we run the script with the patch on Debian Jessie (inside a Docker container). Here is the Dockerfile:
```
FROM debian:jessie
MAINTAINER Evgeny Golyshev <eugulixes@gmail.com>

ENV HOME /root

RUN echo  deb http://ftp.ru.debian.org/debian jessie main    > /etc/apt/sources.list \
 && echo  deb http://emdebian.org/tools/debian/ jessie main >> /etc/apt/sources.list \
 && echo "APT::Default-Release \"stable\";" > /etc/apt/apt.conf

RUN dpkg --add-architecture armhf

RUN apt-get clean && apt-get update && apt-get install -y \
    curl \
 && apt-get install -y --force-yes \
    git \
    libpq-dev \
    locales \
    python3 \
    python3-pip \
    supervisor \

    # Dependencies for rpi2-gen-image
    bc \
    binfmt-support \
    debootstrap \
    dosfstools \
    bmap-tools \
    gcc-arm-linux-gnueabihf \
    psmisc \
    qemu-user-static \
    sudo \
    whois \
 && apt-get clean \
 && rm -rf /var/lib/apt/lists/*

RUN mkdir $HOME/.ssh
COPY id_rsa $HOME/.ssh/id_rsa
COPY id_rsa.pub $HOME/.ssh/id_rsa.pub

RUN ssh-keyscan bitbucket.org > $HOME/.ssh/known_hosts \
 && cd \
 && pip3 install --upgrade pip \
 && git clone git@bitbucket.org:cusdeb/dominion.git \
 && cd dominion \
 && pip install -r requirements.txt \
 && python3 setup.py install

RUN mkdir /var/dominion \
 && cd /var/dominion \
 && git clone -b dominion https://github.com/denismosolov/rpi2-gen-image.git

VOLUME ["/var/dominion/workspace"]

COPY ./dominion/config/supervisord.conf /etc/supervisor/supervisord.conf
COPY ./dominion/run_dominion.sh /usr/bin/run_dominion.sh
COPY ./dominion/run_worker.sh /usr/bin/run_worker.sh
```

We are trying to add a web UI to the script https://cusdeb.com/

drtyhlpr, 2017-01-06:
wow gj :) maybe I can add a reference to it in the readme. I want to update the "Build dependencies" saying that the bootstrapping was tested on minimal jessie and stretch linux systems together with the crossbuild-essential-armhf toolchain. Building on other Linux build systems or using other compiler toolchains is of course possible too but was not tested.

the problem is my time - i dont have mutch of it currently. I even tought about adding support for other boards:
Arietta G25
Aria G25
Acqua A5
Beaglebones
even x86-arm64 why not ? :)
(because the have a very up-to-date kernel-patch github). I decided not to do it because the constant testing and patching "little changes" I do not have continuesly have time for)

drtyhlpr, 2017-01-06:
https://cusdeb.com/ nice idea and page !

some time ago there was a little hype on german it news pages:
http://www.pibakery.org/ - looks nice but very colorfull - maybe you can get some inspiration. (never tested it)

drtyhlpr, 2017-01-10:
crossbuild-essential-armhf: information added to readme

closed 2017-01-10 by drtyhlpr