drtyhlpr/rpi23-gen-image #58: Wrong defaults for ENABLE_ROOT and ENABLE_ROOT_SSH

shaardie, 2016-11-04:
According to the Readme `ENABLE_ROOT` and `ENABLE_ROOT_SSH` should have default `false`, but `rpi23-gen-image.sh` shows:
    
    ENABLE_ROOT=${ENABLE_ROOT:=false}
    ENABLE_ROOT_SSH=${ENABLE_ROOT_SSH:=false}

denismosolov, 2016-11-14:
Hello, 
I've noticed a similar issue and created a pull request here: https://github.com/drtyhlpr/rpi23-gen-image/pull/55
But I'm not sure what should: README.md or rpi23-gen-image.sh
@shaardie what do you think

shaardie, 2016-11-14:
I would prefer
ENABLE_ROOT=${ENABLE_ROOT:=true}
ENABLE_ROOT_SSH=${ENABLE_ROOT_SSH:=false}

since these are the defaults in Debian Jessie.

drtyhlpr, 2017-01-05:
I fixed the readme. I did not change the defaults.

closed 2017-01-05 by drtyhlpr