drtyhlpr/rpi23-gen-image #59: Dependencies missing

shaardie, 2016-11-04:
For using the scripts I had to install the following extra dependencies under Debian Jessie:

 * psmisc 
 * build-essential (I guess for gcc and libc6-dev)
 * bc
 * dbus

gfa, 2016-12-06:
i had to install device-tree-compiler to be able to build the kernel (rpi 3)

drtyhlpr, 2017-01-05:
thanks for the info will patch soon. sorry for the delay

drtyhlpr, 2017-01-06:
hm I tested it today. Running Debian jessie from netinstall with only "essential system tools" installed inside vmware.

jessie:
psmisc: should be a jessie default package
build-essential: should not be required
bc -> added with patch
dbus: should be a jessie default package

the "crossbuild-essential-armhf" was removed from jessie and emdedian repo has to be added:
echo "deb http://emdebian.org/tools/debian/ jessie main" > /etc/apt/sources.list.d/crosstools.list
```
sudo -u nobody wget -O - http://emdebian.org/tools/debian/emdebian-toolchain-archive.key | apt-key add -
dpkg --add-architecture armhf
apt-get update
```

ps. 
stretch:
bc -> just added with patch
psmisc -> just added with patch
rest seems to work out of the box. testing running releases soon rpi2 - rpi3

drtyhlpr, 2017-01-06:
@gfa hi, can you tell me what command lines you used for device tree compilation. are the default ones included in the rpi2/3 kernel sources not working (any more)? or did you have a problem with a specific dev driver?

drtyhlpr, 2017-01-06:
ah you are using UBOOT. well I tried it some months ago and failed to get the device up and running with it.  (dark/lsdcolor screen on all configuration tries) -> https://github.com/drtyhlpr/rpi23-gen-image/issues/47.

so with using "apt-get install device-tree-compiler" (or adding it to the APT required packages) it was/is working? sorry I don't have any rpi around and will be able to test next weekend.

gfa, 2017-01-06:
Later down the road I found #47, I removed uboot  from the options and installed device-tree-compiler and the image worked fine 😀

Do you want me to test without uboot if the problem appears again? I didn't test without uboot and without device-tree-compiler

I don't have a pi to test but I can test the image creation

drtyhlpr, 2017-01-06:
hehe same here... of course all tests and results from your side would be nice. but I think I am just doing exatly the same on some slower vmware hosts. (building on jessie and stretch - added device-tree-compiler etc.).. hitting the console every 30min and doing other so called work in between :)

what currently worries me the most is: https://github.com/drtyhlpr/rpi23-gen-image/issues/61 - I had this problems before as described and invested alot of time to debug and stuff ... without success. problem automatically solved 1-2 weeks later... maybe I got a RPi3 available next to me on Sunday .

drtyhlpr, 2017-01-10:
seems fixed. testing with various parameter combinations the next days

closed 2017-01-10 by drtyhlpr