drtyhlpr/rpi23-gen-image #65: Dri2

Greg798100, 2017-01-08:
Hi. Perfectly everything is compiled. But at start of kodi constantly I have libEGL warning error: DRI2: failed to authenticate. Tried ubunu mate, arch linux, raspbian and there I have no such mistakes.

drtyhlpr, 2017-01-09:
hi please tell me details. what raspberry version and what linux release, or what command line did you use? ty

Greg798100, 2017-01-16:
Raspberry 3 B. Crosscompile to virtual Debian jessie SMP Debian 3.16.39-1.
Your script of assembly is good, this the best for me and my purposes. And Kodi has replaced with omxplayer+youitube-dl+cromium-browser (launchpad.net/ubuntu). Thanks, question is closed.

zerkman, 2017-01-16:
Same issue on RPi 2. Installing libraspberrypi0 does get libEGL.so and other installed in /opt/vc/lib.

zerkman, 2017-01-16:
Same issue here on RPi 2, with libraspberrypi0 installed, which enables all libs in /opt/vc.

Seems Kodi needs to be built with specific Raspberry Pi settings, and the version from the Debian package is not.

drtyhlpr, 2017-01-23:
https://github.com/xbmc/xbmc/blob/master/docs/README.raspberrypi

hmmm

Greg798100, 2017-01-23:
Thanks. It won't be enough life to compile. I tried and had mistakes, but not on this readme. It was curious to solve problems with vcgencmd. I understood have it not one.  I think problem with kodi and omxplayer in xorg the driver.  As it https://packages.debian.org/sid/x11/x11proto-dri2-dev and http://elinux.org/RPi_Xorg_rpi_Driver.

closed 2017-08-21 by drtyhlpr