drtyhlpr/rpi23-gen-image #66: RPi3 with Stretch does not boot

drtyhlpr, 2017-01-10:
The current stretch release does currently not boot on RPi3 (hangs after WLAN detection, no matter if firmware available or not). If you have ideas how to fix it - feel free to help

g-vidal, 2017-01-12:
I came across this problem once (with previous script version), I had not been able to trace out the cause but it seemed that the raspi was not able to boot without proper wifi config, this did not happen with wired connections. I copied wifi informations from my laptop to the wpa_supplicant.conf  file and it was OK but that's more a workaround than a solution...sorry

drtyhlpr, 2017-01-25:
hm ok - i tried some stuff about the wireless configuration but it did not work... hmm...

@gfa I think you said you got it working - can you paste your command line parameters?

gfa, 2017-01-26:
i only have one rpi3 running stretch and wireless, and i'm just getting started using wireless so i dont know how stable is it
wireless works very reliably in jessie (should be the same, same kernel)

```shell
# Configuration template file used by rpi23-gen-image.sh
# HW
RPI_MODEL=3
RELEASE=stretch
BUILD_KERNEL=true
ENABLE_UBOOT=true
ENABLE_CONSOLE=true
ENABLE_INITRAMFS=false
# build
APT_PROXY="http://127.0.0.1:3142/"
ENABLE_MINBASE=false
KERNELSRC_DIR=/var/tmp/linux
KERNELSRC_CLEAN=false
KERNELSRC_CONFIG=true
ENABLE_REDUCE=false #otherwise dropbear will be used
ENABLE_MINGPU=true
#APT_INCLUDES="wpasupplicant"
REDUCE_DOC=true
REDUCE_MAN=true
CHROOT_SCRIPTS="XXXXX/rpi23-gen-image/scripts/"
# sys
APT_SERVER=deb.debian.org
ENABLE_HWRANDOM=true
ENABLE_SOUND=false
TIMEZONE=Etc/UTC
EXPANDROOT=false
ENABLE_NONFREE=true
ENABLE_RSYSLOG=false
# network
ENABLE_HARDNET=false
ENABLE_IPTABLES=false
ENABLE_DHCP=true
ENABLE_IPV6=true
ENABLE_WIRELESS=true
ENABLE_SSHD=true
# user
ENABLE_USER=false
ENABLE_ROOT=true
SSH_DISABLE_PASSWORD_AUTH=true
SSH_ROOT_AUTHORIZED_KEYS="XXXXXX/authorized_keys"
SSH_ENABLE_ROOT=true
ENABLE_ROOT=true
```
the chroot script does this

```
#!/bin/sh

apt-get install -y wpasupplicant avahi-daemon

cat <<EOF > /etc/wpa_supplicant/wpa_supplicant-wlan0.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
        ssid="CeluAP"
        scan_ssid=1
        psk="XXXXXXX"
        key_mgmt=WPA-PSK
        group=CCMP TKIP
        priority=90
}
EOF

systemctl enable wpa_supplicant@wlan0.service

mkdir /etc/systemd/network
cat <<'EOF' > /etc/systemd/network/20-wlan0.conf
[Match]
Name=wlan0

[Link]
# fix the mac address
MACAddress=b8:27:eb:6b:XX:XX

[Network]
DHCP=yes
EOF
```

drtyhlpr, 2017-01-26:
very nice thanks - so I can use this config as basis for my patching.

drtyhlpr, 2017-01-28:
hm does boot with enabled wireless even without wlan config hm. doing some further test but seems to work now. yeah

closed 2017-01-28 by drtyhlpr