drtyhlpr/rpi23-gen-image #67: Serial console currenly does not work with U-Boot (RPi3)

drtyhlpr, 2017-01-10:
working on a fix.  should affect RPi3 only.

drtyhlpr, 2017-01-11:
I am still working with this RPi3 serial console stuff. Its a (little bit) messy.... argh

there is even a presentation about the serial changes available lol:
http://de.slideshare.net/yeokm1/raspberry-pi-3-uartbluetooth-issues

burnbabyburn, 2018-11-19:
I think you got it right. You enabled fastest serial mode (page 13) while bluetooth still works with the following packages:

https://aur.archlinux.org/packages/pi-bluetooth/
or 
https://github.com/RPi-Distro/pi-bluetooth/tree/master/debian

burnbabyburn, 2018-11-24:
Info Links:
- https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/
- https://hallard.me/enable-serial-port-on-raspberry-pi/

Pull is comming

burnbabyburn, 2018-12-07:
still broken under u-boot. Only ttyAMA0 and serial1 are created :/

burnbabyburn, 2018-12-13:
PhilE explained it better here: https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=213499