drtyhlpr/rpi23-gen-image #73: Keyboard settings seems to not working with console-setup on RPI3

supergilbert, 2017-02-06:
The script /bin/setupcon compare date (with test <cached_file> -ot <config_file>) to regenerate the cached files. My raspberry doesn't have a real clock. So the <config_file> is generated in 2014 and <cached_file> is newer.
(My solution or temporay hack is to remove cached files to force the regeneration in a custom script)
(I do not look at keyboard under X11)

drtyhlpr, 2017-02-08:
hm I currently don't know why there should be a file created in 2014 if your build system is running the correct data. Can you please add the commandline or config template that you used - so I can use this as a basis to perform the same steps like you did. of course no rpi hardware has  realtime clock on it but it should be possible to add a simple i2c based one ... and yeah this nasty big battery.

supergilbert, 2017-02-09:
Sorry I was not able to reproduce my problem. 
Maybe I have made a wrong configuration or I was too much influenced by people that have problems to update date, when they are in "2014" (The default date on my RPI3 without any setting).
If I see the problem again. I willi try to bring you all usefull the information.
But for now you can consider that as a mistake.

drtyhlpr, 2017-02-09:
hi no problem. I ll leave this issue open for a couple of weeks so maybe another user can give message if he has same or similar problems. to be honest I haven't tested the keyboard lang setting much :)

supergilbert, 2017-02-13:
It seems to stay some problems with the keyboard configuration under the console.

The keyboard configuration have a strong dependency with packages "keyboard-configuration" and "locales".

In the script "bootstrap.d/12-locale.sh" line 13. There is a grep on ",locales". That seems to not work in some cases.
 
With the folowing configuration:
RPI_MODEL=3
APT_INCLUDES=locales
XKB_MODEL="pc105"
XKB_LAYOUT="fr"
XKB_VARIANT="latin9"
XKB_OPTIONS=""

I have got:

> 
> ...
> + + grep ,locales
> echo locales,apt-transport-https,apt-utils,ca-certificates,debian-archive-keyring,dialog,sudo,systemd,sysvinit-utils,parted,dbus,openssh-server,alsa-utils,rng-tools
> + [ ]
> ...
> 

What follow the echo is the variable APT_INCLUDES
The keyboard configuration does not seems to be done.

(At line 43 we have got "chroot_exec dpkg-reconfigure -f noninteractive keyboard-configuration", but I do not know if this package is installed without including it in the variable APT_INCLUDES)

drtyhlpr, 2017-02-19:
hi thanks for the info. sorry for the delay in the reply. I will test and patch it

drtyhlpr, 2017-02-22:
not fixed yet. the required locales (and keyboard) packages are only installed if DEFLOCAL is different from en_US.

```
# Add required locales packages
if [ "$DEFLOCAL" != "en_US.UTF-8" ] ; then
  APT_INCLUDES="${APT_INCLUDES},locales,keyboard-configuration,console-setup"
fi
```

I thought the XKB_ settings are only relevant for the xorg keyboard settings, but https://wiki.debian.org/Keyboard reads different,

burnbabyburn, 2018-11-10:
i always included these packages in apt_include. (German language didn't work without it) seems like this is the same error or atleast related.

I think this should fix the problem:

```
if [ "$DEFLOCAL" != "en_US.UTF-8" ] || ([ -n XKB_MODEL ] || [ -n XKB_LAYOUT ] ||  [ -n XKB_VARIANT ] ||  [ -n XKB_OPTIONS ]); then
  APT_INCLUDES="${APT_INCLUDES},locales,keyboard-configuration,console-setup"
fi
```

PR #155 (2018-11-15): Fixes
- iptables path fix
- crosscompiler check fix
-fix #73 
-fix RPI0 not including WIFI Firmware for MODEL=0W
- firmware blob link in readme fix
- iptables need iptables-persistent package to provide 

due to the fact i made soo many errors, i'll compile every PR i do...

**BTW**:  I would suggest using kernel version 4.14 as default atm. Latest rpi-firmware [commits](https://github.com/raspberrypi/firmware/commit/55e591283f456ab9cebd9c31aaef4939f814880d) are for this version and kernel 4.18 has issues with [bpf_syscall](https://github.com/drtyhlpr/rpi23-gen-image/issues/145#issuecomment-43 [... cut]
closed 2018-11-15 by drtyhlpr