drtyhlpr/rpi23-gen-image #76: Generating Debian packages for kernel upgrades

459below, 2017-03-24:
I'm running the script regulary to compile new kernel images. It's not too much of a hassle to install the new images and modules by hand, however it is kind of slow and tedious. Generating Debian packages would make this step faster and more reliable. I reckon it to be a rather simple task to implement this functionality. If I successfully implement this, I will send an PR request. Unless of course somebody else does it first or points out that it is already done somewhere.

459below, 2017-05-05:
Just to be clear on this. My suggestion is to extend the functionality of rpi23-gen-image with a switch to generate .deb packages in addition to the image itself.

I got this idea from this project https://github.com/scus1/dns320l
This can generate .deb kernel and kernel module packages for the SOC used by D-Link's NAS 320L.

JRWR, 2017-05-05:
thats what I was going to do anyway, using the built in scripts already to build the chroot and spit out some debs

459below, 2017-07-21:
Appending "deb-pkg" to https://github.com/drtyhlpr/rpi23-gen-image/blob/master/bootstrap.d/13-kernel.sh#L97 spits out several .deb kernel packages. Most interestingly linux-image-4.9.38-v7+_4.9.38-v7+-1_armhf.deb and linux-headers-4.9.38-v7+_4.9.38-v7+-1_armhf.deb. I will look into as if they are complete.

Furthermore as they weigh in with 11 MB and 16 MB my first guess would be, that it would be reasonable to enable this by default.

459below, 2017-07-21:
The image package seems to contain everything it should. However one would need to copy the kernel to /boot/firmware and rename it or modify config.txt respectively.

Unfortunately the headers don't seem to work as they should. Though they are put in the right spot for DKMS to pick up, it will fail on the actual compilation with:

`/bin/sh: 1: scripts/basic/fixdep: Exec format error`

drtyhlpr, 2017-08-21:
hi 459below - great work !1 - i will investiage and fix the kernel headers issue (they are installed by the script but the install location is not detected correctly due to chroot - if i remember right)

drtyhlpr, 2017-08-21:
hi, not 100% sure if this is related to the issue #87 - symlinks should now be correct.

459below, 2017-08-21:
Hello drtyhlpr, I'm building the image right now. I'm going to take a look at how the kernel-header .deb package comes out.

Regarding this issue itself I cared about a convenient way to build and install kernel images on a already installed Debian HF. Adding "deb-pkg" to the make command is such that. I will open a PR in which the fitness of this change can be discussed more explicitly.

drtyhlpr, 2017-08-21:
i agree having .deb kernel files that could "simply" be generated and installed would be great. I am still a bit bussy with other stuff but I wanted to add a "generate qemu kernel" option (like https://github.com/dhruvvyas90/qemu-rpi-kernel/blob/master/tools/build-kernel-qemu) in some days

ps. hehe this patch looks simple :)

drtyhlpr, 2017-08-21:
`/bin/sh: 1: scripts/basic/fixdep: Exec format error`

this error is a bit strange and sounds like wrong arch exec. (so script, fixdep wrong arch = not arm, or a arm exec called from build system arch (amd64 fe)

459below, 2017-08-21:
Yes, that's what I thought, too. I have elaborated on this over at the PR thread https://github.com/drtyhlpr/rpi23-gen-image/pull/98

drtyhlpr, 2017-12-25:
sadly does not work :(

closed 2017-12-25 by drtyhlpr