drtyhlpr/rpi23-gen-image #83: Crash with error message on ping

g-vidal, 2017-06-07:


quozl, 2017-06-07:
I'm not surprised.  Stretch is currently a moving target, and may occasionally not build.  Either;

* wait until the problem is fixed by Debian developers, or
* go back in time to a previous set of Stretch packages, using http://snapshot.debian.org/ in your sources, or
* reproduce the problem with Debian tools alone (such as debootstrap and not rpi23-gen-image) and report it to Debian, or
* diagnose and fix the problem yourself.

You might also look at a fork of rpi23-gen-image to see if it fares any differently; see https://github.com/michaelfranzl/rpi23-gen-image

brianmay, 2017-06-19:
I built stretch fine yesterday (with just some minor issues needing fixing afterward), so maybe this is fixed.

drtyhlpr, 2017-08-21:
hi, thanks for the feedback. i need some days till i have access to my rpi. no pis at my current location :( will test and fix is required.

drtyhlpr, 2017-08-21:
i tested different build configurations today and the problem should be fixed (by debian)

closed 2017-08-21 by drtyhlpr