drtyhlpr/rpi23-gen-image #86: Raspberry Pi 3 Stretch fails: Couldn't execute /usr/bin/apt-key

brianmay, 2017-06-28:
```
+ [ -z  ]
+ install_readonly files/apt/10proxy /home/brian/rpi23-gen-image/images/stretch/build/chroot/etc/apt/apt.conf.d/10proxy
+ install -o root -g root -m 644 files/apt/10proxy /home/brian/rpi23-gen-image/images/stretch/build/chroot/etc/apt/apt.conf.d/10proxy
+ sed -i s/""/""/ /home/brian/rpi23-gen-image/images/stretch/build/chroot/etc/apt/apt.conf.d/10proxy
+ [ true = false ]
+ install_readonly files/apt/sources.list /home/brian/rpi23-gen-image/images/stretch/build/chroot/etc/apt/sources.list
+ install -o root -g root -m 644 files/apt/sources.list /home/brian/rpi23-gen-image/images/stretch/build/chroot/etc/apt/sources.list
+ sed -i s/\/ftp.debian.org\//\/localhost:9999\// /home/brian/rpi23-gen-image/images/stretch/build/chroot/etc/apt/sources.list
+ sed -i s/ jessie/ stretch/ /home/brian/rpi23-gen-image/images/stretch/build/chroot/etc/apt/sources.list
+ [ false = true ]
+ chroot_exec apt-get -qq -y update
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /home/brian/rpi23-gen-image/images/stretch/build/chroot apt-get -qq -y update
W: GPG error: http://localhost:9999/debian stretch-updates InRelease: Couldn't execute /usr/bin/apt-key to check /var/lib/apt/lists/partial/localhost:9999_debian_dists_stretch-updates_InRelease
W: The repository 'http://localhost:9999/debian stretch-updates InRelease' is not signed.
W: GPG error: http://localhost:9999/debian stretch Release: Couldn't execute /usr/bin/apt-key to check /var/lib/apt/lists/partial/localhost:9999_debian_dists_stretch_Release
W: The repository 'http://localhost:9999/debian stretch Release' is not signed.
W: GPG error: http://security.debian.org stretch/updates InRelease: Couldn't execute /usr/bin/apt-key to check /var/lib/apt/lists/partial/security.debian.org_dists_stretch_updates_InRelease
W: The repository 'http://security.debian.org stretch/updates InRelease' is not signed.
+ chroot_exec apt-get -qq -y -u dist-upgrade
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /home/brian/rpi23-gen-image/images/stretch/build/chroot apt-get -qq -y -u dist-upgrade
WARNING: The following packages cannot be authenticated!
  libc6 libc-bin multiarch-support libgnutls30 libexpat1
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...
```

Still trying to work out why it cannot execute /usr/bin/apt-key

quozl, 2017-06-28:
Because Debian Stretch changed to run apt-key as user not root.  Is the same as https://github.com/drtyhlpr/rpi23-gen-image/pull/85, try the patch there?

brianmay, 2017-06-28:
Thanks. Will try. I searched existing issues before hand, however (yet again) forgot to check the pull requests.

brianmay, 2017-06-28:
Seems to be working now. Is currently cloning the kernel source. Am I bit puzzled though how I generated Raspberry Pi 2 stretch images without any problems.

quozl, 2017-06-28:
Glad it is working.  I'm not puzzled; rpi23-gen-image relies on inputs that are subject to change, it is both a design advantage and disadvantage. Discovering the change would require preserving information about the state of those inputs.  If you have build logs, you might diff them.  Changing rpi23-gen-image to be idempotent would be difficult; easiest would be to configure it to use local inputs only.

drtyhlpr, 2017-08-21:
(c&p) should be fixed. testing now various build settings to make sure no build errors exist.

closed 2017-08-21 by drtyhlpr