drtyhlpr/rpi23-gen-image #87: installed kernel headers installed - where?

brianmay, 2017-06-29:
According to my reading of the documentation, kernel headers should be installed by default. Important, as I want to try an alternative wifi driver, see if it solves my problems with my RTL8188CUS wifi adapter. https://github.com/pvaret/rtl8192cu-fixes

Unfortunately, I can't actually find the kernel headers. The closest I have come is this:

```
root@rpi2-stretch:~# ls -l /lib/modules/4.9.33-v7+/{build,source}
lrwxrwxrwx 1 root root 69 Jun 19 19:41 /lib/modules/4.9.33-v7+/build -> /home/brian/rpi23-gen-image/images/stretch/build/chroot/usr/src/linux
lrwxrwxrwx 1 root root 69 Jun 19 19:41 /lib/modules/4.9.33-v7+/source -> /home/brian/rpi23-gen-image/images/stretch/build/chroot/usr/src/linux
```

Which is the references to the location on the build system. I was kind of hoping that "KERNEL_HEADERS=true" which I believe is the default would mean the header files are installed somewhere...

quozl, 2017-06-29:
Yes, you're right, the build and source symlinks have leaked the build system paths.  Same here.  The built image does not have any kernel headers.  https://github.com/drtyhlpr/rpi23-gen-image/blob/master/bootstrap.d/13-kernel.sh#L117 should do it, perhaps it is doing it to the build system instead of the built system.

459below, 2017-07-21:
Building kernel modules is an issue I've been having, too. My guess is that my issue addresses the same problem and the fix seems extremely simple. As the kernel should be updated regularly anyway, I ignored the fact, that kernel-headers wouldn't be installed initially.
https://github.com/drtyhlpr/rpi23-gen-image/issues/76

drtyhlpr, 2017-08-21:
hi *, I tought there was a patch or at least a comment about someone with the same problem (kernel headers are installed at wrong location , ~chroot get path wrong~), hm I tought there was a pr/commit with a manual cp -r if I remember right. I will test and fix as soon as possible.

drtyhlpr, 2017-08-21:
I am currently doing some test with INSTALL_MOD_PATH of modules_install to investigate the problem. "relative" INSTALL_HDR_PATH="../.." seams to work fine and the headers should be in the correct location.

drtyhlpr, 2017-08-21:
Hi! , the latest commit: https://github.com/drtyhlpr/rpi23-gen-image/commit/376da6a50b03148169bf63ced309aa0d84dbd5e3

should fix this problem. (so build and source are now symlinked to /usr/src/linux).

please give some feedback if this now works as expected. hf!

closed 2017-10-28 by drtyhlpr