drtyhlpr/rpi23-gen-image #88: Debian buster image ?

g-vidal, 2017-06-29:
Hi all,
It has been difficult to migrate from jessie to stretch but it seems that migration to buster is very smooth... Would-it be possible to easily build a debian buster image ? Is it possible to check if it works just by changing  stretch to buster in some files?
Thanks

svlog, 2017-07-07:
it works perfect i've tested and created a debian buster(arm64) image without issues for rpi3 at least the image... 

if you wanna test it i can upload the image for you.

g-vidal, 2017-07-14:
Thanks for the info and the proposal :-)

I will build one by myself and confirm here

g-vidal, 2017-07-16:
Hi,
I ams sorry but it does not work at all.
I had to do  amendmendments on various script files as described in the discussions here, (for systemd-sysv exclusion, _apt new configuration...)  but the build always fails with or without additionnal packages. 
My last attempt crashes like this:
`+ chroot_exec /bin/bash -x
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /data/RpiGenImage/Images/buster/build/chroot /bin/bash -x
+ cd /tmp/xf86-video-fbturbo
+ autoreconf -vi
/bin/bash: line 2: autoreconf: command not found
+ ./configure --prefix=/usr
/bin/bash: line 3: ./configure: No such file or directory
+ make
make: *** No targets specified and no makefile found.  Stop.
+ make install
make: *** No rule to make target 'install'.  Stop.
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...
`
And it is very annoying to start again the full process instead of starting back from the crash point.
Sorry for that bad news, I can try any proposal.
Thanks again

svlog, 2017-07-16:
It looks like a dependency problem.. have you installed all required packages? be sure you've got all including autoreconf.

g-vidal, 2017-07-17:
I did some more amendments with dependencies as you say, but it still fails with this one that I cannot see how to cure... :
```shell
./configure: line 18655: syntax error near unexpected token `RANDR,'
./configure: line 18655: `XORG_DRIVER_CHECK_EXT(RANDR, randrproto)'
+ make
make: *** No targets specified and no makefile found.  Stop.
+ make install
make: *** No rule to make target 'install'.  Stop.
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...
```
Thanks for your quick answers, I am stuck  but I can try  some ideas if you have more.

svlog, 2017-07-17:
I haven't tested to create the image with xorg support.. I think this error is caused by the xorg-video-fbturbo, so is not related by rpi32-gen-image
try disabling ENABLE_XORG in config and once you've create the image successfully add xorg directly in your rpi.

g-vidal, 2017-07-17:
Hi,
I have tried with and without fbturbowith no effect  but I agreen it might be the culprit. 
I have disabled FBTURBO, _WM and ENABLE_XORG. We'll see but this  is not very convenient as it was sooooo comfortable to have an X11 raspi out of the box. :-)

I am also investigating this track : [RANDR: syntax error near unexpected token](http://marc.info/?l=freedesktop-xorg&m=117278341437072) that may help?

svlog, 2017-07-17:
Ok, still a dependency problem. I tested and now is compiling. Install xserver-xorg-dev xorg-dev and not sure(xorg-server-source) and should compiling fine.

g-vidal, 2017-07-18:
Thanks it works !
I have even removed --allow-unauthenticated as recommended on #85

svlog, 2017-07-18:
glad to hear that!

g-vidal, 2017-07-27:
This discussion can be closed I have made all the amendments available on   [my branch(https://github.com/g-vidal/rpi23-gen-image) and added some **verified** templates (maybe with some redundant packages in the list that cause warnings). They correspond to my needs, but it can be of some interest as examples. if needed I can make a pull request  but I am not sure thatthe community will need/want all that stuff.

closed 2017-07-27 by g-vidal