drtyhlpr/rpi23-gen-image #89: rpi23-gen-image fails with GPG errors

ljones0, 2017-07-06:
Hello! Unfortunatly I'm having some problems with trying to use rpi23-gen-image on a debain buster system (btw, I have also tried rpi23-gen-image on a stretch system too but I get the same problem). Here's the problem, or rather the output of the rpi23-gen-image command:

(System is an x86-64,  memory size=8GB, CPU=AMD Phenom II x4 955, debian)

+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /home/aaaaa/1/rpi23-gen-image-master/images/stretch/build/chroot apt-get -qq -y update
W: GPG error: http://security.debian.org stretch/updates InRelease: Couldn't execute /usr/bin/apt-key to check /var/lib/apt/lists/partial/security.debian.org_dists_stretch_updates_InRelease
W: The repository 'http://security.debian.org stretch/updates InRelease' is not signed.
W: GPG error: http://ftp.debian.org/debian stretch-updates InRelease: Couldn't execute /usr/bin/apt-key to check /var/lib/apt/lists/partial/ftp.debian.org_debian_dists_stretch-updates_InRelease
W: The repository 'http://ftp.debian.org/debian stretch-updates InRelease' is not signed.
W: GPG error: http://ftp.debian.org/debian stretch Release: Couldn't execute /usr/bin/apt-key to check /var/lib/apt/lists/partial/ftp.debian.org_debian_dists_stretch_Release
W: The repository 'http://ftp.debian.org/debian stretch Release' is not signed.
+ chroot_exec apt-get -qq -y -u dist-upgrade
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /home/aaaaa/1/rpi23-gen-image-master/images/stretch/build/chroot apt-get -qq -y -u dist-upgrade
WARNING: The following packages cannot be authenticated!
  libc6 libc-bin multiarch-support libgcrypt20 libgnutls30 libexpat1
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...

For some reason there are some packages which can't be authenticated. I've tried running apt-get update but it made no difference; apt-key update just says "Warning: 'apt-key update' is deprecated and should not be used anymore! Note: In your distribution this command is a no-op and can therefore be removed safely." -- so apt-key update isn't possible.

does anyone know what could cause this problem?

ljones

ljones0, 2017-07-06:
oop, my mistake. Someone beat me to it, mine I think is similar to #86....

svlog, 2017-07-06:
Workaround 

https://github.com/drtyhlpr/rpi23-gen-image/pull/85/commits/2e44635ee733953953f8c745ba4a4fe0f9b44154

svlog.

drtyhlpr, 2017-08-21:
should be fixed. testing now various build settings to make sure no build errors exist.

closed 2017-08-21 by drtyhlpr