drtyhlpr/rpi23-gen-image #94: back for testing and fixing

drtyhlpr, 2017-08-19:
hi users and supporters, sorry for beeing away for so long time ... bussy real life.

so I am moving back to debian and my rpis became quite a bit dusty... hm well I am testing the latest versions for compile errors etc and try to find time for this project on a 2 week basis or such....including pulls etc

@Bobberty Thanks for the amd64 patches etc. looking forward to test in merge soon

quozl, 2017-08-19:
Welcome back.

Have a look at the repository forks, there's one in particular with some considerable change you may be interested in borrowing or cherry-picking from.  I know I was.  ;-)

-- 
James Cameron
http://quozl.netrek.org/

drtyhlpr, 2017-08-21:
hi thanks for the cherries :), looks nice - as_nobody, i ll do some merge changes for the (sudo -E) pull merged yesterday and merge

closed 2017-10-28 by drtyhlpr