drtyhlpr/rpi23-gen-image #100: No 'Hardware' in /proc/cpuinfo for RPi3/aarch64/Stretch

koolkhel, 2017-09-22:
Hello! I'm using RPi3/aarch64/Stretch/kernel-4.11 with a slightly customized rpi23-gen-image (I set scripts to boot from USB root). I'd like to use GPIO handling code, but it seems that any related software (like wiringPi, RPi.GPIO etc) doesn't recognize my RPi3.

root@rpi3-stretch:~# python test_v1.py
Traceback (most recent call last):
  File "test_v1.py", line 3, in <module>
    import RPi.GPIO as GPIO
  File "/usr/local/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!

I found the most likely cause to be /proc/cpuinfo not reporting hardware ID:
root@rpi3-stretch:~# cat /proc/cpuinfo
processor	: 0
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

And that seems crucial for related code.

What are available options when we know the cause? It seems like a slightly wrong kernel configuration here, but I'm not sure.

koolkhel, 2017-10-05:
The trouble seems linked with https://github.com/raspberrypi/linux/issues/2110 so it is a generic Linux Kernel bug for this 64-bit architecture.

burnbabyburn, 2018-10-27:
Sadly the bug got a won't fix, so this issue can be closed

drtyhlpr, 2018-10-28:
:-1:

closed 2018-10-28 by drtyhlpr

burnbabyburn, 2019-10-12:
This [Post](https://lb.raspberrypi.org/forums/viewtopic.php?t=214315)
Links to
[Sakakis patch](https://github.com/sakaki-/genpi64-overlay/blob/master/dev-embedded/wiringpi/wiringpi-2.44-r7.ebuild)

1. download wiring pi _- its depricated now btw_
2. apply `pseudo-cpuinfo.patch` from sakaki
3. copy `cpuinfo `from repo to `/etc/wiringpi/cpuinfo`
4. compile
5. profit

433Mhz Tools didnt compile...