drtyhlpr/rpi23-gen-image #108: debootstrap fails on arm64 buster

imigueldiaz, 2017-12-03:
Hi,
It is not a failure on scripts, but I do know where else to ask, debootstrap second stage always fails with:

W: Failure while unpacking required packages.  This will be attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package /var/cache/apt/archives/bash_4.4-5_arm64.deb is at fault)

debootstrap relevant info says:

pkg: warning: ignoring pre-dependency problem!
Preparing to unpack .../archives/bash_4.4-5_arm64.deb ...
preinst: ../sysdeps/unix/sysv/linux/spawni.c:366: __spawnix: Assertion `ec >= 0' failed.
qemu: uncaught target signal 6 (Aborted) - core dumped
dpkg: error processing archive /var/cache/apt/archives/bash_4.4-5_arm64.deb (--unpack):
 new bash package pre-installation script subprocess was killed by signal (Aborted)

It seems to be a problem on bash package on arm64 architecture, my template is like:

RPI_MODEL=3
RELEASE=buster
BUILD_KERNEL=true
RELEASE_ARCH=arm64
KERNEL_ARCH=arm64
HOSTNAME=hostname
PASSWORD=XXXXX
USER_PASSWORD=yyyyy
XKBMODEL=pc105
XKBLAYOUT=es
ENABLE_DHCP=false
NET_ADDRESS=X.X.X-X/ZZ
NET_GATEWAY=Y.Y.Y.Y
NET_DNS_1=8.8.8.8
NET_DNS_2=8.8.4.4
NET_DNS_3=Z.Z.Z.Z
ENABLE_NONFREE=true
ENABLE_WIRELESS=true

Could you give any direction about possible workarounds or where to report the problem?

Best Regards
Ignacio

drtyhlpr, 2017-12-04:
Hi thanks for your feedback and thanks for the detailed configuration you used. I will investigative the problem in a couple of days and find a solution :)

drtyhlpr, 2017-12-06:
hi @imigueldiaz please check https://github.com/Microsoft/WSL/issues/1878. looks quite the same hmmm.
``
sudo apt-mark hold bash
``
``
sudo apt-mark hold ebtables
``
may be a temporary solution (hm what shell whould be installed instead idk).

imigueldiaz, 2017-12-07:
Hi!

Thanks both for your fast and accurate answers! Finally I have managed to succesfully create an Stretch arm64 image, and I'm playing with It :) 

Based on @g-vidal  answer I saw that non-english users must include on APT_INCLUDES_LATE packages as console-data, locales, keyboard-configuration because without them the keyboard remains in US or GB and it is a bit traumatic for Spanish users at first :P.

I solved it by installing them manually after first login with apt install and dpkg-reconfigure with no problem at all (my image was already generated). 

When I have more time, I'll try to experiment with buster image on a spare SD card and I'll inform you about results ;)

Best regards,
Ignacio

closed 2017-12-07 by imigueldiaz