drtyhlpr/rpi23-gen-image #109: I build buster with xfce4 failed.

youxiaojie, 2017-12-19:
dpkg: error processing package librest-0.7-0:armhf (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 dconf-service
 xfconf
 libxfconf-0-2
 thunar
 libthunarx-2-0
 xfce4-appfinder
 xfce4-settings
 xfce4-panel
 dconf-gsettings-backend:armhf
 xfce4
 libxfce4ui-1-0
 xfdesktop4
 gsettings-desktop-schemas
 libgtk-3-common
 xfce4-pulseaudio-plugin:armhf
 libexo-1-0:armhf
 xfwm4
 exo-utils
 xfce4-session
 libxfce4ui-2-0
 libgtk-3-0:armhf
 libxfce4ui-utils
 libgarcon-1-0
 glib-networking:armhf
 libxfce4panel-2.0-4
 libkeybinder-3.0-0:armhf
 libsoup2.4-1:armhf
 libsoup-gnome2.4-1:armhf
 librest-0.7-0:armhf

drtyhlpr, 2017-12-23:
hi, urg! - I hope we can blame the buster release state for that one. please try again in 1-2 weeks. please keep in mind that buster is the current debian "testing" release. testing is not stable and broken packages happen from time to time in this release state.

drtyhlpr, 2017-12-25:
does the problem still exits?

closed 2018-02-17 by drtyhlpr