drtyhlpr/rpi23-gen-image #110: how to updater kernel?

youxiaojie, 2017-12-20:
how to updater kernel? should I need compiled from source again?

drtyhlpr, 2017-12-23:
hi youxiaojie! yes if you want to update the kernel you will need to build/compile it from the source aggain. Debian does not offer a binary rpi2/3 kernel (that you simple could fetch via apt upgrade. in general it is also possible to build the kernel on a running rpi2/3 plattform but so called cross compiling is the prefered way because its so mutch faster!

To understand more about the kernel building process please take a look at:
https://www.raspberrypi.org/documentation/linux/kernel/building.md

closed 2017-12-25 by drtyhlpr

youxiaojie, 2017-12-25:
compile kernel in rpi is ok,make install or so.
if i cross compile kernel in pc, how to make install ? do as script did?
firstly chroot to sdcard, 
secondly  run make?
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcm2709_defconfig
make  ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- zImage modules dtbs
make  ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=../../.. modules_install
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_FW_PATH=../../../lib firmware_install
make  ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_HDR_PATH=../.. headers_install
and then?