drtyhlpr/rpi23-gen-image #111: why is there no contrib component?

youxiaojie, 2017-12-23:
# Use non-free Debian packages if needed
if [ "$ENABLE_NONFREE" = true ] ; then
  COMPONENTS="main,non-free"
fi

drtyhlpr, 2017-12-23:
hi, because since 3 years noone asked for it :-) non-free is not contrib (i am sure you know). if there is a good reason to include contrib (even if its only due to your personal need) just tell me and I can add it ... or add it urself the script is pritty structured and adding a ENABLE_CONTRIB option by 90% copy and paste of ENABLE_NONFREE ... there you go :)

drtyhlpr, 2017-12-24:
ah sorry now i see the error. yes contrib is added to apt sources at file 11-apt.sh but not in the 10-bootsrap.sh. fixed. thanks for report.

closed 2017-12-24 by drtyhlpr