drtyhlpr/rpi23-gen-image #112: why use ":=" ?

youxiaojie, 2017-12-23:
what does := mean? as in
 PASSWORD=${PASSWORD:=raspberry}

drtyhlpr, 2017-12-23:
hi!

the ":=" operator is used in this /bin/sh script on almost any configuration parameter variable to set a default value if the parameter is not set or empty.
  
${parameter:=default} - means - if parameter not set, set it to value of default.

details: http://www.tldp.org/LDP/abs/html/parameter-substitution.html

closed 2017-12-25 by drtyhlpr

youxiaojie, 2017-12-25:
the ":=" operator is used in this /bin/sh script on almost any configuration parameter variable to set a default value if the parameter is not set or empty.

${parameter:=default} - means - if parameter not set, set it to value of default.

details: http://www.tldp.org/LDP/abs/html/parameter-substitution.html

thanks.
is it only for bash?

quozl, 2017-12-25:
Some other shells handle `:=`.  Since `bash` is a default shell on some systems and not others, over years scripts end up having such _bashisms_ removed after problems are reported.  So if you detect this, best is to make a pull request;
* if the script is not `source`d, change the first line to refer to `/bin/bash`, or;
* replace the use of `bash`-unique feature with a comparable POSIX feature.

drtyhlpr, 2018-01-03:
hi! my goal was to create a simple "dependency free" (yeah that changes over the time ...) sh script in the spirit of good old /etc/*, init.d fashion. I thought that no matter what default shell may be used on the build linux system /bin/sh will be there for sure. of course i agree that bash is std on many many linux systems. sh seamed more "pure" - lol maybe that was an illusion (* i still love u bash /dev/tcp/hrhr)