drtyhlpr/rpi23-gen-image #114: error of .config file not found. Solved

youxiaojie, 2017-12-25:
#
# Build and Setup RPi2/3 Kernel
#
+ . bootstrap.d/13-kernel.sh
+ . ./functions.sh
+ [ true = true ]
+ mkdir -p /home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux
+ [ -n /home/allan/rpi_linux/linux ]
+ [ -d /home/allan/rpi_linux/linux ]
+ cp -r /home/allan/rpi_linux/linux/COPYING /home/allan/rpi_linux/linux/CREDITS /home/allan/rpi_linux/linux/Documentation /home/allan/rpi_linux/linux/Kbuild /home/allan/rpi_linux/linux/Kconfig /home/allan/rpi_linux/linux/MAINTAINERS /home/allan/rpi_linux/linux/Makefile /home/allan/rpi_linux/linux/Module.symvers /home/allan/rpi_linux/linux/README /home/allan/rpi_linux/linux/System.map /home/allan/rpi_linux/linux/arch /home/allan/rpi_linux/linux/block /home/allan/rpi_linux/linux/built-in.o /home/allan/rpi_linux/linux/certs /home/allan/rpi_linux/linux/crypto /home/allan/rpi_linux/linux/drivers /home/allan/rpi_linux/linux/firmware /home/allan/rpi_linux/linux/fs /home/allan/rpi_linux/linux/include /home/allan/rpi_linux/linux/init /home/allan/rpi_linux/linux/ipc /home/allan/rpi_linux/linux/kernel /home/allan/rpi_linux/linux/lib /home/allan/rpi_linux/linux/mm /home/allan/rpi_linux/linux/modules.builtin /home/allan/rpi_linux/linux/modules.order /home/allan/rpi_linux/linux/net /home/allan/rpi_linux/linux/samples /home/allan/rpi_linux/linux/scripts /home/allan/rpi_linux/linux/security /home/allan/rpi_linux/linux/sound /home/allan/rpi_linux/linux/tools /home/allan/rpi_linux/linux/usr /home/allan/rpi_linux/linux/virt /home/allan/rpi_linux/linux/vmlinux /home/allan/rpi_linux/linux/vmlinux.o /home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux
+ [ false = true ]
+ [ 1 = 1 ]
+ [ -r /proc/cpuinfo ]
+ grep -c processor /proc/cpuinfo
+ KERNEL_THREADS=2
+ [ true = false ]
+ [ ! -r /home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux/arch/arm/boot/zImage ]
+ [ false = true ]
+ make -C /home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_MOD_PATH=../../.. modules_install
make: Entering directory '/home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux'
  SHIPPED scripts/kconfig/zconf.tab.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
scripts/kconfig/Makefile:37: recipe for target 'silentoldconfig' failed
make[2]: *** [silentoldconfig] Error 1
Makefile:547: recipe for target 'silentoldconfig' failed
make[1]: *** [silentoldconfig] Error 2
  DEPMOD  4.13.16-v7+
make: Leaving directory '/home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux'
+ make -C /home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- INSTALL_FW_PATH=../../../lib firmware_install
make: Entering directory '/home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux'
scripts/kconfig/conf  --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
scripts/kconfig/Makefile:37: recipe for target 'silentoldconfig' failed
make[2]: *** [silentoldconfig] Error 1
Makefile:547: recipe for target 'silentoldconfig' failed
make[1]: *** [silentoldconfig] Error 2
  IHEX    firmware/acenic/tg1.bin
  INSTALL ...lib/acenic/tg1.bin
  IHEX    firmware/acenic/tg2.bin
  INSTALL ...lib/acenic/tg2.bin
  IHEX    firmware/adaptec/starfire_rx.bin
  INSTALL ...lib/adaptec/starfire_rx.bin
  IHEX    firmware/adaptec/starfire_tx.bin
  INSTALL ...lib/adaptec/starfire_tx.bin
  IHEX    firmware/dsp56k/bootstrap.bin
  INSTALL ...lib/dsp56k/bootstrap.bin
  IHEX2FW firmware/atmsar11.fw
  INSTALL ...lib/atmsar11.fw
  IHEX    firmware/bnx2x/bnx2x-e1-6.2.9.0.fw
  INSTALL ...lib/bnx2x/bnx2x-e1-6.2.9.0.fw
  IHEX    firmware/bnx2x/bnx2x-e1h-6.2.9.0.fw
  INSTALL ...lib/bnx2x/bnx2x-e1h-6.2.9.0.fw
  IHEX    firmware/bnx2x/bnx2x-e2-6.2.9.0.fw
  INSTALL ...lib/bnx2x/bnx2x-e2-6.2.9.0.fw
  IHEX    firmware/bnx2/bnx2-mips-09-6.2.1a.fw
  INSTALL ...lib/bnx2/bnx2-mips-09-6.2.1a.fw
  IHEX    firmware/bnx2/bnx2-rv2p-09-6.0.17.fw
[... 267 lines cut ...]
  INSTALL include/linux/tc_act (15 files)
  INSTALL include/linux/tc_ematch (4 files)
  INSTALL include/linux/usb (11 files)
  INSTALL include/linux/wimax (1 file)
  INSTALL include/misc (1 file)
  INSTALL include/mtd (5 files)
  INSTALL include/rdma (19 files)
  INSTALL include/rdma/hfi (2 files)
  INSTALL include/scsi (4 files)
  INSTALL include/scsi/fc (4 files)
  INSTALL include/sound (15 files)
  INSTALL include/video (3 files)
  INSTALL include/xen (4 files)
  INSTALL include/asm (37 files)
make: Leaving directory '/home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux'
+ mkdir /home/allan/rpi23-gen-image/working/images/buster/build/chroot/boot/firmware
+ cat /home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux/include/config/kernel.release
+ KERNEL_VERSION=4.13.16-v7+
+ install_readonly /home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux/.config /home/allan/rpi23-gen-image/working/images/buster/build/chroot/boot/config-4.13.16-v7+
+ install -o root -g root -m 644 /home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux/.config /home/allan/rpi23-gen-image/working/images/buster/build/chroot/boot/config-4.13.16-v7+
install: cannot stat '/home/allan/rpi23-gen-image/working/images/buster/build/chroot/usr/src/linux/.config': No such file or directory
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...

youxiaojie, 2017-12-25:
commit here:
https://github.com/drtyhlpr/rpi23-gen-image/pull/113

drtyhlpr, 2017-12-25:
thanks for the great patch. did not notice that problem during testing @youxiaojie can you please move your changes of the README.md (Add Simplified Chinese Translation) to README-CN.md so I can join your pull request easily.

closed 2017-12-25 by drtyhlpr