drtyhlpr/rpi23-gen-image #115: no reboot halt man...commad?

youxiaojie, 2017-12-25:
no reboot halt man command....

drtyhlpr, 2017-12-25:
hi please provide more info. what release and what is the exact error message? like I wrote before take care with problems on buster release it is not stable - it is testing (like temporary broken xfce4). does rebooting via systemctl work?

```
systemctl poweroff
systemctl reboot
```

youxiaojie, 2017-12-25:
commonly we use reboot and halt command to reboot and power off.but not include. man command is also.

drtyhlpr, 2017-12-25:
this seams strange and man, reboot and halt are definitely available on images generated with jessie and stretch releases. I am not sure if this is a tempory state of buster or if they more and more push to uniq systemctl control.

nice "problem description":
https://askubuntu.com/questions/631929/reboot-and-shutdown-executables-missing

drtyhlpr, 2017-12-25:
so if you search the packages for the "reboot" binary:
https://packages.debian.org/search?suite=buster&arch=any&searchon=contents&keywords=reboot

this should be included in:
https://packages.debian.org/buster/systemd-sysv

so maybe you just need to install the package. (if there are no other package related problems)

closed 2018-02-17 by drtyhlpr