drtyhlpr/rpi23-gen-image #119: xfce4 could not be installed

youxiaojie, 2017-12-30:
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package xfconf is at fault)
xfce concerned package could not be configured.please test.

drtyhlpr, 2017-12-31:
hi @youxiaojie. are you talking about the buster debian release ? If you try to build buster release please notice what I wrote before about the "testing state" of debian buster. (and read about it on the debian homepage) broken packages can occur quite often of debian buster release and a possibly solution may only be to disable the package temporary.

ps. if I got u wrong and you are not talking about the buster release please give me some pasts of error logs. thanks

youxiaojie, 2017-12-31:
no newly installed stretch

youxiaojie, 2017-12-31:
to build a stretch image under newly installed stretch debian

Script started on Sat 30 Dec 2017 10:25:58 PM CST
-e 
#
# RPi2/3 Bootstrap Settings
#
+ RPI_MODEL=2
+ RPI2_DTB_FILE=bcm2709-rpi-2-b.dtb
+ RPI2_UBOOT_CONFIG=rpi_2_defconfig
+ RPI3_DTB_FILE=bcm2710-rpi-3-b.dtb
+ RPI3_UBOOT_CONFIG=rpi_3_32b_defconfig
+ RELEASE=stretch
+ KERNEL_ARCH=arm
+ RELEASE_ARCH=armhf
+ CROSS_COMPILE=arm-linux-gnueabihf-
+ COLLABORA_KERNEL=3.18.0-trunk-rpi2
+ [ arm = arm64 ]
+ KERNEL_DEFCONFIG=bcm2709_defconfig
+ KERNEL_IMAGE=kernel7.img
+ [ armhf = arm64 ]
+ QEMU_BINARY=/usr/bin/qemu-arm-static
+ KERNEL_BRANCH=rpi-4.13.y
+ KERNEL_URL=https://github.com/raspberrypi/linux
+ FIRMWARE_URL=https://github.com/raspberrypi/firmware/raw/master/boot
+ WLAN_FIRMWARE_URL=https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm80211/brcm
+ COLLABORA_URL=https://repositories.collabora.co.uk/debian
+ FBTURBO_URL=https://github.com/ssvb/xf86-video-fbturbo.git
+ UBOOT_URL=git://git.denx.de/u-boot.git
+ pwd
+ BASEDIR=/home/allan/rpi23-gen-image/working/images/stretch
+ BUILDDIR=/home/allan/rpi23-gen-image/working/images/stretch/build
+ date +%Y-%m-%d
+ DATE=2017-12-30
+ [ -z rpi-4.13.y ]
+ IMAGE_NAME=/home/allan/rpi23-gen-image/working/images/stretch/2017-12-30-arm-rpi-4.13.y-rpi2-stretch-armhf
+ R=/home/allan/rpi23-gen-image/working/images/stretch/build/chroot
+ ETC_DIR=/home/allan/rpi23-gen-image/working/images/stretch/build/chroot/etc
+ LIB_DIR=/home/allan/rpi23-gen-image/working/images/stretch/build/chroot/lib
+ BOOT_DIR=/home/allan/rpi23-gen-image/working/images/stretch/build/chroot/boot/firmware
+ KERNEL_DIR=/home/allan/rpi23-gen-image/working/images/stretch/build/chroot/usr/src/linux
+ WLAN_FIRMWARE_DIR=/home/allan/rpi23-gen-image/working/images/stretch/build/chroot/lib/firmware/brcm
+ RPI_FIRMWARE_DIR=
+ HOSTNAME=rpi2-stretch
+ PASSWORD=raspberry
+ USER_PASSWORD=raspberry
+ DEFLOCAL=en_US.UTF-8
+ TIMEZONE=Europe/Berlin
+ EXPANDROOT=true
+ XKB_MODEL=
+ XKB_LAYOUT=
+ XKB_VARIANT=
+ XKB_OPTIONS=
+ ENABLE_DHCP=true
+ NET_ADDRESS=
+ NET_GATEWAY=
+ NET_DNS_1=
+ NET_DNS_2=
+ NET_DNS_DOMAINS=
+ NET_NTP_1=
+ NET_NTP_2=
+ APT_PROXY=
+ APT_SERVER=mirrors.tuna.tsinghua.edu.cn
+ ENABLE_CONSOLE=true
+ ENABLE_I2C=false
+ ENABLE_SPI=false
+ ENABLE_IPV6=true
+ ENABLE_SSHD=true
+ ENABLE_NONFREE=false
+ ENABLE_WIRELESS=false
[... 1831 lines cut ...]
I: Configuring tasksel...
I: Configuring libcryptsetup4:armhf...
I: Configuring libparted2:armhf...
I: Configuring parted...
I: Configuring systemd...
I: Configuring libc-bin...
I: Configuring ca-certificates...
I: Configuring libgdk-pixbuf2.0-0:armhf...
I: Configuring dbus...
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package xfconf is at fault)
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package xfconf is at fault)
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package xfconf is at fault)
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package xfconf is at fault)
W: Failure while configuring base packages.  This will be re-attempted up to five times.
W: See //debootstrap/debootstrap.log for details (possibly the package xfconf is at fault)
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...

Script done on Sat 30 Dec 2017 10:54:27 PM CST

drtyhlpr, 2018-01-01:
hi, oh ok thanks for the info. I just tried myself with a minimal configuration and xfce. so if you read the bootstrap.log it smells like this is the problem:

> Package default-dbus-session-bus is not installed.
> Package dbus-session-bus is not installed.

without further investigation i would say: thank systemd for breaking it (aggain and aggain and aggain...)... well sarcasm of - i will find a solution. the blog post is quite usefull.

https://sugaronraspberrypi.wordpress.com/2017/06/23/much-awaited-blog-post/

thanks again for reporting

drtyhlpr, 2018-01-01:
fixed. thanks for reporting

https://github.com/drtyhlpr/rpi23-gen-image/commit/e09e278fb2ce1075db77ae3e53ab59e7a098ad69

quozl, 2018-01-01:
Interesting, thanks.  I'm the Quozl referred to in that blog post.

Opinion; at heart, the problem may be your use of `--no-install-recommends`.  Great for keeping images small, but not so great because Debian dependencies are not fully described.  A package may rely on recommends to satisfy some of the functions it has, and not others.  Debian packagers don't test their packages with `--no-install-recommends`, so whenever you use it things keep breaking like this.

drtyhlpr, 2018-01-03:
oh... ups thanks for the hint ... i ll take a closer look on that :+1:

youxiaojie, 2018-01-03:
I have looked into and try to find something, but fails.
next time I want to try mate desktop, to see anything will happen.
last I try mate-desktop-environt (meta package) fails, seem like this problems too.

drtyhlpr, 2018-01-03:
well in general I want this script to create small images (one of the design goals of the script, back then there was no rasberian-lite available). if you want a full desktop environment (that may finally even get to the size of a full blown desktop) the rasberian-lite may be a alternative solution. (don't get me wrong I don't want to scare you away).

youxiaojie, if you are building a default stretch debian image then flash and boot it and then try the apt-get install mate-desktop-environment (manually). does this work ? 

xfce4 is build and installing with the latest patch. please tell me what sub packages fail on mate-desktop meta AND please please include detailed error logs! (using the "insert code function" may even make it more readable).

so if you read the history of broken packages in the closed issues of this script - you will not find many reported problems (that are not related to "testing" releases) - in my opinion the script is running great for many many users (and sub projects, image generator webpages etc) based on all kind of package installations all based on --no-install-recommend (or even minbase, --no-install-recommend etc)

drtyhlpr, 2018-01-03:
here this should fix your problem:

```
RPI_MODEL=3
RELEASE=stretch
BUILD_KERNEL=true
ENABLE_WM=xfce4
APT_INCLUDES=mate-desktop-environment,debian-mate-default-settings,mate-session-manager
```

if i find time I maybe find a better solution for checking for package dependencies. if you have other package installation problems also consider testing its with the APT_INCLUDES_LATE parameter.

youxiaojie, 2018-01-03:
I agree with you,smaller is better. The reason that I like Debian and Arch is they are all install a base system and install what users want gradually.
I just wish to test mate with your script if it is ok. I think it is ok when I install mate by apt command under raspberry.
 Regarding to  xfce4,mentioned by you in readme ,"supported", so I must test if it is work:)

thanks

youxiaojie, 2018-01-03:
I think your are right. Let the system running in raspberry to deal with more complicated dependency.
For base system The easier the better. support xfce4 as you said in Readme is enough.
So I just to test mate and learn something from your script. thanks.

drtyhlpr, 2018-01-03:
ok :) good luck. sorry if I may sound a bit harsh but broken packages and debian are a very sensitive topic for me. problems with the intel (backport etc) packages etc made me leave debian 1-2y ago after wasting tons of time (and intel gfx on xorg problems are even in some old childhood memories LOL was that 18y ago ? omg) ..., I may use debian packaging in the non default way - based on minbase and --no-install.recommend ... but i went into alot of horror keeping my system small (and getting all attempts "destroyed" .. by "in my point of view" very stupid set default package dependencies).

youxiaojie, 2018-01-03:
What brand of linux are you using ? What is your mind to Arch linux?

drtyhlpr, 2018-01-03:
yup arch :+1:  but i never used it on a raspberry !

youxiaojie, 2018-01-03:
why?  why not on raspberry pi? I see Arch has a most widely embeded devices. nas, single board , and pogo plug etc.....

yup arch👍but i never used it on a raspberry !

drtyhlpr, 2018-01-03:
hi, most of my embedded projects (1,2 y ago) were smaller arm based devices (BLE arm chips etc.) - I did some development with riotOS. I came to raspberry because I needed more power (opencv camera stuff). So to be honest I havent touched any embedded device or board since about 10months because I am bussy with other projects.

so no specific reason I did not use ARCH on raspberry, I just did not give it a try yet (yeah there was a big push to embedded also looking at ubuntu (or its sub projects) etc.

closed 2018-02-17 by drtyhlpr