drtyhlpr/rpi23-gen-image #120: No more firmware_install if building with Kernel 4.14

g-vidal, 2018-01-03:
Hi all,
There has been a change in kernel 4.14 !
`firmware_install` is not any more available in the kernel build if you use 4.14 from github
rpi23-gen-image build is crashing in `bootstrap.d/k13-kernel.sh`    at that line 
```shell
# Install kernel firmware no more present in Kernel 4.14                                                                                                                                   
    make -C "${KERNEL_DIR}" ARCH="${KERNEL_ARCH}" CROSS_COMPILE="${CROSS_COMPILE}" INSTALL_FW_PATH=../../../lib firmware_install      
```        
@drtyhlpr I don't know how to handle best this condition, to have clean scripts...

drtyhlpr, 2018-01-03:
hi g-vidal thanks for the report .... i will search for a solution tomorrow etc!

drtyhlpr, 2018-01-04:
g-vidal do you have any homepage link that references to this change? yeah i googled etc. but can not realy find an announcement. (yeah i will reproduce the bug... but frustrating to not easily find a google hit on that important change).

g-vidal, 2018-02-13:
here are the addresses I found on that topic : 

```
http://lkml.iu.edu/hypermail/linux/kernel/1709.1/04650.html
http://rglinuxtech.com/?p=2049
```
I have not been able to find the original message said to be at 
`git://git.kernel.org/pub/scm/linux/kernel/git/gregkh/driver-core.git/ tags/firmware_removal-4.14-rc1`
 I just found : `https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/driver-core.git/commit/?id=5620a0d1aacd554ebebcff373e31107bb1ef7769`

Hope that helps

keopx, 2018-02-20:
g-vidal I solved changing this:

```diff
diff --git a/bootstrap.d/13-kernel.sh b/bootstrap.d/13-kernel.sh
index b488592..896c102 100644
--- a/bootstrap.d/13-kernel.sh
+++ b/bootstrap.d/13-kernel.sh
@@ -115,7 +115,7 @@ if [ "$BUILD_KERNEL" = true ] ; then
     make -C "${KERNEL_DIR}" ARCH="${KERNEL_ARCH}" CROSS_COMPILE="${CROSS_COMPILE}" INSTALL_MOD_PATH=../../.. modules_install
 
     # Install kernel firmware
-    make -C "${KERNEL_DIR}" ARCH="${KERNEL_ARCH}" CROSS_COMPILE="${CROSS_COMPILE}" INSTALL_FW_PATH=../../../lib firmware_install
+    make -C "${KERNEL_DIR}" ARCH="${KERNEL_ARCH}" CROSS_COMPILE="${CROSS_COMPILE}" INSTALL_FW_PATH=../../../lib
   fi
 
   # Install kernel headers

```

Based on https://patchwork.kernel.org/patch/9960627/

drtyhlpr, 2018-02-20:
hi g-vidal, thanks for tracing and fixing this. well as long as there are no missing firmware binary blobs I am very happy to have less of them :+1:  Ill try to create a small "IF KERNEL_VERSION > 4.14... sed the firmware_install DONE"-patch this weekend if I find time. thx again

drtyhlpr, 2018-09-27:
fixed. finally!

closed 2018-09-27 by drtyhlpr