drtyhlpr/rpi23-gen-image #121: '/debootstrap/debootstrap': Exec format error

amanharitsh123, 2018-02-02:
I am using debian in a schroot environment on ubuntu 16.04 .There is some issue with the building the image , I am providing the logs here 

`Script started on Fri Feb  2 20:23:44 2018
(debian)root@amanharitsh-Lenovo-G50-70:/home/amanharitsh/Desktop/Rpi/rpi23-gen-im
mage# exitrm -r images 
mage# CONFIG_TEMPLATE=rpi3stretch  ./rpi23-gen-image.sh
-e 
#
# RPi2/3 Bootstrap Settings
#
+ RPI_MODEL=3
+ RPI2_DTB_FILE=bcm2709-rpi-2-b.dtb
+ RPI2_UBOOT_CONFIG=rpi_2_defconfig
+ RPI3_DTB_FILE=bcm2710-rpi-3-b.dtb
+ RPI3_UBOOT_CONFIG=rpi_3_32b_defconfig
+ RELEASE=stretch
+ KERNEL_ARCH=arm
+ RELEASE_ARCH=armhf
+ CROSS_COMPILE=arm-linux-gnueabihf-
+ COLLABORA_KERNEL=3.18.0-trunk-rpi2
+ [ arm = arm64 ]
+ KERNEL_DEFCONFIG=bcm2709_defconfig
+ KERNEL_IMAGE=kernel7.img
+ [ armhf = arm64 ]
+ QEMU_BINARY=/usr/bin/qemu-arm-static
+ KERNEL_BRANCH=
+ KERNEL_URL=https://github.com/raspberrypi/linux
+ FIRMWARE_URL=https://github.com/raspberrypi/firmware/raw/master/boot
+ WLAN_FIRMWARE_URL=https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm80211/brcm
+ COLLABORA_URL=https://repositories.collabora.co.uk/debian
+ FBTURBO_URL=https://github.com/ssvb/xf86-video-fbturbo.git
+ UBOOT_URL=git://git.denx.de/u-boot.git
+ pwd
+ BASEDIR=/home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch
+ BUILDDIR=/home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build
+ date +%Y-%m-%d
+ DATE=2018-02-02
+ [ -z  ]
+ IMAGE_NAME=/home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/2018-02-02-arm-CURRENT-rpi3-stretch-armhf
+ R=/home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot
+ ETC_DIR=/home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot/etc
+ LIB_DIR=/home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot/lib
+ BOOT_DIR=/home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot/boot/firmware
+ KERNEL_DIR=/home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot/usr/src/linux
+ WLAN_FIRMWARE_DIR=/home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot/lib/firmware/brcm
+ RPI_FIRMWARE_DIR=
+ HOSTNAME=rpi3-stretch
+ PASSWORD=raspberry
+ USER_PASSWORD=raspberry
+ DEFLOCAL=en_US.UTF-8
+ TIMEZONE=Europe/Berlin
+ EXPANDROOT=true
+ XKB_MODEL=
+ XKB_LAYOUT=
+ XKB_VARIANT=
+ XKB_OPTIONS=
+ ENABLE_DHCP=true
+ NET_ADDRESS=
+ NET_GATEWAY=
+ NET_DNS_1=
+ NET_DNS_2=
+ NET_DNS_DOMAINS=
+ NET_NTP_1=
+ NET_NTP_2=
+ APT_PROXY=
+ APT_SERVER=ftp.debian.org
+ ENABLE_CONSOLE=true
+ ENABLE_I2C=false
+ ENABLE_SPI=false
+ ENABLE_IPV6=true
[... 566 lines cut ...]
I: Extracting libblkid1...
I: Extracting libfdisk1...
I: Extracting libmount1...
I: Extracting libsmartcols1...
I: Extracting libuuid1...
I: Extracting mount...
I: Extracting util-linux...
I: Extracting liblzma5...
I: Extracting zlib1g...
+ install -m 755 -o root -g root /usr/bin/qemu-arm-static /home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot/usr/bin/qemu-arm-static
+ mkdir -p /home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot/usr/share/keyrings
+ install_readonly /usr/share/keyrings/debian-archive-keyring.gpg /home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot/usr/share/keyrings/debian-archive-keyring.gpg
+ install -o root -g root -m 644 /usr/share/keyrings/debian-archive-keyring.gpg /home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot/usr/share/keyrings/debian-archive-keyring.gpg
+ chroot_exec /debootstrap/debootstrap --second-stage
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /home/amanharitsh/Desktop/Rpi/rpi23-gen-image/images/stretch/build/chroot /debootstrap/debootstrap --second-stage
chroot: failed to run command '/debootstrap/debootstrap': Exec format error
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...
(debian)root@amanharitsh-Lenovo-G50-70:/home/amanharitsh/Desktop/Rpi/rpi23-gen-im
mage# exit
exit

Script done on Fri Feb  2 20:28:46 2018`

quozl, 2018-02-03:
When I had this kind of problem, I fixed it by using Debian instead of Ubuntu.  I used a Debian schroot inside my Ubuntu system.  I don't know that your problem will be fixed by this, but it is something you can try.

amanharitsh123, 2018-02-03:
Actually , I am also doing the same ,debian schroot inside my Ubuntu 16.04 but still the problem exists.  I will now try to build the image on my another machine running Ubuntu 17.10 . Hope it works this time

amanharitsh123, 2018-02-03:
I have found a  fix i think 
Firstly we have to install binfmt-support on the host system , then "update-binfmts --enable qemu-arm" in the schroot in environemnt , the script successfully created the rpi image after this .

deejoe, 2018-02-03:
See also https://github.com/michaelfranzl/rpi23-gen-image/issues/5

amanharitsh123, 2018-02-03:
well , as i mentioned  here https://github.com/michaelfranzl/rpi23-gen-image/issues/5 in the last comment , I was getting an error  warning: Couldn't load the binfmt_misc module 
It is solved by installing binfmt-support on the host system.