drtyhlpr/rpi23-gen-image #125: Not build rpi3-stretch-arm64-4.11.y image

keopx, 2018-02-20:
I am running this and not create the image:

`# CONFIG_TEMPLATE=rpi3-stretch-arm64-4.11.y ./rpi23-gen-image.sh`

That is a default template.

Current running systems is:

`Linux debian 4.9.0-5-amd64 #1 SMP Debian 4.9.65-3+deb9u2 (2018-01-04) x86_64 GNU/Linux`

And stopped here:

```
# Setup Networking
#
+ . bootstrap.d/20-networking.sh
+ . ./functions.sh
+ install_readonly files/network/hostname /data/rpi23-gen-image/images/stretch/build/chroot/etc/hostname
+ install -o root -g root -m 644 files/network/hostname /data/rpi23-gen-image/images/stretch/build/chroot/etc/hostname
+ sed -i s/^rpi2-jessie/rpi3-stretch/ /data/rpi23-gen-image/images/stretch/build/chroot/etc/hostname
+ install_readonly files/network/hosts /data/rpi23-gen-image/images/stretch/build/chroot/etc/hosts
+ install -o root -g root -m 644 files/network/hosts /data/rpi23-gen-image/images/stretch/build/chroot/etc/hosts
+ sed -i s/rpi2-jessie/rpi3-stretch/ /data/rpi23-gen-image/images/stretch/build/chroot/etc/hosts
+ [  !=  ]
+ [ true = false ]
+ install_readonly files/network/interfaces /data/rpi23-gen-image/images/stretch/build/chroot/etc/network/interfaces
+ install -o root -g root -m 644 files/network/interfaces /data/rpi23-gen-image/images/stretch/build/chroot/etc/network/interfaces
+ install_readonly files/network/eth.network /data/rpi23-gen-image/images/stretch/build/chroot/etc/systemd/network/eth.network
+ install -o root -g root -m 644 files/network/eth.network /data/rpi23-gen-image/images/stretch/build/chroot/etc/systemd/network/eth.network
+ [ true = true ]
+ sed -i -e s/DHCP=.*/DHCP=yes/ -e /DHCP/q /data/rpi23-gen-image/images/stretch/build/chroot/etc/systemd/network/eth.network
+ [ true = false ]
+ sed -i /.*=$/d /data/rpi23-gen-image/images/stretch/build/chroot/etc/systemd/network/eth.network
+ [ stretch = stretch ]
+ mv -v /data/rpi23-gen-image/images/stretch/build/chroot/etc/systemd/network/eth.network /data/rpi23-gen-image/images/stretch/build/chroot/lib/systemd/network/10-eth.network
'/data/rpi23-gen-image/images/stretch/build/chroot/etc/systemd/network/eth.network' -> '/data/rpi23-gen-image/images/stretch/build/chroot/lib/systemd/network/10-eth.network'
+ rm -fr /data/rpi23-gen-image/images/stretch/build/chroot/etc/systemd/network
+ chroot_exec systemctl enable systemd-networkd
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /data/rpi23-gen-image/images/stretch/build/chroot systemctl enable systemd-networkd
Created symlink /etc/systemd/system/multi-user.target.wants/systemd-networkd.service -> /lib/systemd/system/systemd-networkd.service.
Created symlink /etc/systemd/system/sockets.target.wants/systemd-networkd.socket -> /lib/systemd/system/systemd-networkd.socket.
+ install_readonly files/network/host.conf /data/rpi23-gen-image/images/stretch/build/chroot/etc/host.conf
+ install -o root -g root -m 644 files/network/host.conf /data/rpi23-gen-image/images/stretch/build/chroot/etc/host.conf
+ [ false = true ]
+ [ NET_NTP_1 !=  ]
+ chroot_exec systemctl enable systemd-timesyncd.service
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /data/rpi23-gen-image/images/stretch/build/chroot systemctl enable systemd-timesyncd.service
+ [ true = true ]
+ [ ! -d /data/rpi23-gen-image/images/stretch/build/chroot/lib/firmware/brcm ]
+ mkdir -p /data/rpi23-gen-image/images/stretch/build/chroot/lib/firmware/brcm
+ as_nobody mktemp -d
+ sudo -E -u nobody LANG=C LC_ALL=C mktemp -d
+ temp_dir=/tmp/tmp.CBsEvm1Yfc
+ as_nobody wget -q -O /tmp/tmp.CBsEvm1Yfc/brcmfmac43430-sdio.bin https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm80211/brcm/brcmfmac43430-sdio.bin
+ sudo -E -u nobody LANG=C LC_ALL=C wget -q -O /tmp/tmp.CBsEvm1Yfc/brcmfmac43430-sdio.bin https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm80211/brcm/brcmfmac43430-sdio.bin
+ cleanup
+ set +x
killing processes using mount point ...
removing temporary mount points ...
```

I am doing something wrong?

quozl, 2018-02-20:
No, you're doing fine.  The link in the last `wget` is wrong, and the `-q` is hiding the error;

```
Resolving github.com (github.com)... 192.30.255.113, 192.30.255.112
Connecting to github.com (github.com)|192.30.255.113|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-02-21 06:12:15 ERROR 404: Not Found.
```

Correct link is
https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm/brcmfmac43340-sdio.bin

quozl, 2018-02-20:
You are missing https://github.com/drtyhlpr/rpi23-gen-image/commit/dfcf07a6a566aa49a46cf85e2eb785661d8959dd

drtyhlpr, 2018-02-20:
hi keopx, the problem was solved with the updated url (patch), right? ty quozl for ur fast solution :)

keopx, 2018-02-21:
@drtyhlpr Works fine!

Thanks @quozl

closed 2018-02-21 by drtyhlpr