drtyhlpr/rpi23-gen-image #126: Kernel Packages

JRWR, 2018-02-21:
If this project wants to maintain them, I can do weekly packages for rpi2/3 kernels for people to update their Raspberries with

just need the scripts to build and package them

drtyhlpr, 2018-02-22:
Hi JRWR thanks for your support the offer sounds great! Are you talking about deb-pkg/.deb kernel package or are there any other possible ways to do that?

459below and me tried that and failed. maybe you have some ideas, maybe stuff has changed since that:
https://github.com/drtyhlpr/rpi23-gen-image/pull/98

ps. I would love to have rpi23 deb-pkg/.deb kernels!

JRWR, 2018-02-22:
Hrm, I could compile the kernels on hardware, dedicate a scaleway instance to it, since they use the same arch

drtyhlpr, 2018-02-23:
Hi my point was that we tried to use make-kpkg - that should exactly do the "need the script to build and package them". it was tried but (back then) not possible. no matter if a cross_compiler was used or if the package was build inside qemu/bootstrap environment. (https://github.com/drtyhlpr/rpi23-gen-image/pull/98).

I am not very used to create debian packages and I am not very used to make-kpkg. From my basic understanding it just creates a .deb file that includes kernel-source, kernel-image and kernel-headers.

I would not like to maintain, or better be responsible for maintaining binary images (of generated rpi23-$release.img and kernels). But I would love to just a link/ref to the generated kernel packages like the "Weekly image builds" at the end of the README page. I don't know if I find time to include a "EXPORT_KERNEL_PKG" parameter in the script that performs the "make-kpkg" part.

quozl, 2018-02-26:
Linux kernel source has within already scripts for building kernel as a binary package set;

```
make clean
cp foo.config .config
sed -i 's/CONFIG_DEBUG_INFO=y/CONFIG_DEBUG_INFO=n/g' .config
make bindeb-pkg LOCALVERSION=-q0
```

so there's no great need for `make-kpkg` in the Debian `kernel-package` if you have a reason to avoid it.  Disabling `CONFIG_DEBUG_INFO` as above saves much time.  Extending this to cross-compiling requires `CROSS_COMPILE` to be set to the appropriate prefix.  Can be further sped up with `ccache`.

However, it may be more long-term helpful to work with Debian project to have kernels built; the Fedora project does so, after they overcame some licensing issues.  If I recall correctly.

drtyhlpr, 2018-02-27:
@quozl cool - never heard or this `bindeb-pkg` target, thanks for the infos. I ll give it a try as soon as i find time.

459below, 2019-10-03:
So I just skimmed the whole topic here but I want to point out that I've been building kernel deb-packages for a system setup by rpi23-gen-image this whole time without any problems. The script I used for that is [here](https://gist.github.com/459below/8f12904fe70cd7d671a97586f131a668)
Please note that the _cd linux_ refers to the folder in which the **raspberry** linux [sources](https://github.com/raspberrypi/linux) are extracted.
I think something like this would be appropriate to copy the kernel over to the boot partition every time _dpkg -i_ is invoked on the new kernel package:
```
[459below@manus kernel]$ cat /etc/initramfs/post-update.d/zz-update-rpi-kernel 
#!/bin/sh
cp /vmlinuz /initrd.img /boot/firmware/
```
and obviously config.txt would need kernel=vmlinuz and maybe the respective initramfs line if one needs ramfs for encryption or so. Though leaving the old kernel and _sed_ing the _config.txt_ would be more appropriate, obviously. I think the rpi3-firmware package in the Debian repo has something like this.

The other thing is the official Debian support is in the making https://salsa.debian.org/raspi-team/image-specs though my rpi3 runs rpi23-gen with a kernel made by the gist mentioned earlier my rpi2 runs this official debian preview with the official _5.2.0-2-armmp_ kernel from the official Debian repo. A lot of _official_. However, when I tried to get a proper shutdown/power-on GPIO button through the _gpio-shutdown.dtbo_ I noticed that the Debian kernel doesn't seem to have the expected GPIO support. I just learned about this [Debian GPIO project](https://salsa.debian.org/raspi-team/rpi.gpio/blob/master/README.txt), which seems to make "basic" GPIO on the rpi possible, though I suspect it won't be like just setting _dtoverlay=awesome-overlay_ in the config.txt, though.

So I like the idea of having a proper officially supported Debian on the pi with kernel updates and all, but at least GPU and GPIO don't seem to be working as well yet.

459below, 2019-10-03:
OK, so on the image by _gwolf_ I'm still able to swap the kernel with the one I'm running on the rpi23 image, as expected. The thing is that the rpi is so very picky about the DT and fixup stuff. So the Debian kernel is version 5, to which I've not yet upgraded to. Like always: the wifi doesn't want to just work OOTB after the upgrade.

Since I now need a kernel module, hence I'm going to setup a docker build image to produce crosscompiled .deb.

459below, 2019-10-03:
https://hub.docker.com/r/459below/rpi

So I've already build .deb packages with this.

drtyhlpr, 2019-12-09:
sounds good!