drtyhlpr/rpi23-gen-image #127: raspi-config not available?

imranrajjad, 2018-03-21:
hello,

I downloaded the image file directly, after starting the pi 3 I am unable to run raspi-config..I need to configure wifi settings

regards,
Imran

quozl, 2018-03-21:
Normal and expected.  `raspi-config` is not in Debian, but is in Raspbian.  Install the `raspi-config` package from Raspbian.  The program is a user interface for editing various files.  You might also look at what files it changes, and instead change those files.  If you must have the same interfaces as Raspbian then use Raspbian directly.

imranrajjad, 2018-03-22:
I am actually having trouble with installing GNURadio on Raspbian, and am looking for other options, raspi-config is not mandatory but probably working with GPIO pins is.

thanks

quozl, 2018-03-22:
`raspi-config` is a shell script, and is human readable, so run it on Raspbian to find the question you want to answer, then open the script in an editor and search for the question , and you'll find the shell commands that you can run to achieve the same result.  These can be added to a script of your own.  I've looked and all I can see is the "Remote GPIO" feature which enables `pigpiod` as a systemd service.  This comes from the `pigpio` package which isn't in Debian.  I think GPIO access is available without it, it just depends on what API you want to use.

imranrajjad, 2018-04-03:
Ubuntu-mate worked for me

closed 2018-04-03 by imranrajjad