drtyhlpr/rpi23-gen-image #129: U-Boot compilation failed

ioannisky, 2018-04-19:
Hi!

I recently tried to build an image for Raspberry Pi 3 with the following settings:
RPI_MODEL=3
RELEASE=stretch
RELEASE_ARCH=arm64
ENABLE_UBOOT=true
BUILD_KERNEL=false
ENABLE_SPI=true
ENABLE_IPTABLES=true

When U-Boot was compiling I got the following error:
lib/asm-offsets.c:1:0: error: unknown value 'armv5' for -march

The solution was to include gcc-arm-linux-gnueabi in the COMPILER_PACKAGES and to change:
```c
chroot_exec make -j${KERNEL_THREADS} -C /tmp/u-boot/ ${UBOOT_CONFIG} all
```
to
```c
echo "#!/bin/bash" > ${R}/tmp/uboot.sh
echo "export CROSS_COMPILE=arm-linux-gnueabi-" >> ${R}/tmp/uboot.sh
echo "make -j${KERNEL_THREADS} -C /tmp/u-boot/ ${UBOOT_CONFIG} all" >> ${R}/tmp/uboot.sh
chmod u+x ${R}/tmp/uboot.sh
chroot_exec /bin/bash -c /tmp/uboot.sh
```

The image was build successfully but I have not check it yet.
It seems that when the build is for arm64 it requires cross compiler to build the U-boot 32-bits

drtyhlpr, 2018-10-31:
hi, sorry for the late reply. I see a problem in your config template. you need to set BUILD_KERNEL=true this is mandatory for RPI3. I just fixed some uboot stuff and I will do some a64 testing

burnbabyburn, 2018-11-06:
Maybe make BUILD_KERNEL=true as a default? 
BUT
```
if [ "$RELEASE" = "stretch" ]  || [ "$RELEASE" = "buster" ] ; then
  BUILD_KERNEL=true
fi
``` from rpi23-gen.sh should alwys enable BUILD_KERNEL
The error above is more likely from not defining KERNEL_ARCH and BUILD_ARCH

drtyhlpr, 2018-11-09:
hi thanks for your answers.

yeah. back at the time when this script was created many peoples loved to use the precompiled rpi2 kernel and BUILD_KERNEL=true should become default now.

I haven't tested amd64 so far and this is why this text is currently in the readme: "Raspberry Pi 3 images are generated for 32-bit mode only. Raspberry Pi 3 64-bit images can be generated using custom configuration parameters (templates/rpi3-stretch-arm64-4.11.y).".

I am very bussy with other projects but will try to test and fully include amd64 support soon.

burnbabyburn, 2018-11-19:
Following is tested and working on RPI3 and RPi3B+:

- U-Boot 32bit with kernel 32bit
- U-Boot 32bit with kernel 64bit
- U-Boot 64bit with kernel 64bit

drtyhlpr, 2018-11-21:
ok not sure if still problems with qemu. I will close and reopen a new issue for uboot qemu if needed. thanks for your help

closed 2018-11-21 by drtyhlpr