drtyhlpr/rpi23-gen-image #131: Rysnc out of space when making image

JRWR, 2018-05-02:
```
export ENABLE_IPV6=true
export RELEASE="buster"
export PASSWORD="raspberry"
export USER_PASSWORD="raspberry"
export BUILD_KERNEL=true
export ENABLE_I2C=true
export ENABLE_SPI=true
export ENABLE_SSHD=true
export ENABLE_NONFREE=true
export ENABLE_RSYSLOG=true
export ENABLE_SOUND=true
export ENABLE_HWRANDOM=true
export ENABLE_DBUS=true
export ENABLE_IPTABLES=true
export ENABLE_MINGPU=true
```

```
#
# Reduce system disk usage
#
+ . bootstrap.d/99-reduce.sh
+ . ./functions.sh
+ [ false = true ]
+ [ -d custom.d ]
+ [ -n  ]
+ chroot_remove_cc
+ [ ! -z  ]
+ [ buster = jessie ]
+ dbus-uuidgen
+ MACHINE_ID=38d54ad33a798751c9197e215ae9e5ca
+ echo -n 38d54ad33a798751c9197e215ae9e5ca
+ echo -n 38d54ad33a798751c9197e215ae9e5ca
+ chroot_exec apt-get -y clean
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /root/rpi23-gen-image/images/buster/build/chroot apt-get -y clean
+ chroot_exec apt-get -y autoclean
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /root/rpi23-gen-image/images/buster/build/chroot apt-get -y autoclean
Reading package lists...
Building dependency tree...
+ chroot_exec apt-get -y autoremove
+ LANG=C LC_ALL=C DEBIAN_FRONTEND=noninteractive chroot /root/rpi23-gen-image/images/buster/build/chroot apt-get -y autoremove
Reading package lists...
Building dependency tree...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ umount -l /root/rpi23-gen-image/images/buster/build/chroot/proc
+ umount -l /root/rpi23-gen-image/images/buster/build/chroot/sys
+ rm -rf /root/rpi23-gen-image/images/buster/build/chroot/run/*
+ rm -rf /root/rpi23-gen-image/images/buster/build/chroot/tmp/*
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/etc/ssh/ssh_host_*
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/etc/dropbear/dropbear_*
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/etc/apt/sources.list.save
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/etc/resolvconf/resolv.conf.d/original
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/etc/*-
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/etc/apt/apt.conf.d/10proxy
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/etc/resolv.conf
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/root/.bash_history
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/var/lib/urandom/random-seed
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/initrd.img
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/vmlinuz
+ rm -f /root/rpi23-gen-image/images/buster/build/chroot/usr/bin/qemu-arm-static
+ + du -sawk /root/rpi23-gen-image/images/buster/build/chroot
 { print $1 }
+ expr 265100
+ CHROOT_SIZE=265100
+ expr 1 * 1024 * 1024 / 512
+ TABLE_SECTORS=2048
+ expr 64 * 1024 * 1024 / 512
+ FRMW_SECTORS=131072
+ expr 2048 + 131072
+ ROOT_OFFSET=133120
+ expr 265100 + 265100 / 100 * 25
+ expr 331375 * 1024 / 512
+ ROOT_SECTORS=662750
+ expr 2048 + 131072 + 662750
+ IMAGE_SECTORS=795870
+ [ false = true ]
+ dd if=/dev/zero of=/root/rpi23-gen-image/images/buster/2018-05-02-arm-CURRENT-rpi3-buster-armhf.img bs=512 count=2048
2048+0 records in
2048+0 records out
1048576 bytes (1.0 MB, 1.0 MiB) copied, 0.0148968 s, 70.4 MB/s
+ dd if=/dev/zero of=/root/rpi23-gen-image/images/buster/2018-05-02-arm-CURRENT-rpi3-buster-armhf.img bs=512 count=0 seek=795870
0+0 records in
0+0 records out
0 bytes copied, 0.000124455 s, 0.0 kB/s
+ sfdisk -q -L -uS -f /root/rpi23-gen-image/images/buster/2018-05-02-arm-CURRENT-rpi3-buster-armhf.img
+ losetup -o 1M --sizelimit 64M -f --show /root/rpi23-gen-image/images/buster/2018-05-02-arm-CURRENT-rpi3-buster-armhf.img
+ FRMW_LOOP=/dev/loop0
+ losetup -o 65M -f --show /root/rpi23-gen-image/images/buster/2018-05-02-arm-CURRENT-rpi3-buster-armhf.img
+ ROOT_LOOP=/dev/loop1
+ [ false = true ]
+ mkfs.vfat /dev/loop0
mkfs.fat 4.1 (2017-01-24)
+ mkfs.ext4 /dev/loop1
mke2fs 1.44.1 (24-Mar-2018)
Discarding device blocks: done
Creating filesystem with 331372 1k blocks and 82984 inodes
Filesystem UUID: 12d32cc3-6220-43fc-a833-3773b0e71327
Superblock backups stored on blocks:
        8193, 24577, 40961, 57345, 73729, 204801, 221185

Allocating group tables: done
Writing inode tables: done
Creating journal (8192 blocks): done
Writing superblocks and filesystem accounting information: done

+ mkdir -p /root/rpi23-gen-image/images/buster/build/mount
+ mount /dev/loop1 /root/rpi23-gen-image/images/buster/build/mount
+ mkdir -p /root/rpi23-gen-image/images/buster/build/mount/boot/firmware
+ mount /dev/loop0 /root/rpi23-gen-image/images/buster/build/mount/boot/firmware
+ rsync -a /root/rpi23-gen-image/images/buster/build/chroot/ /root/rpi23-gen-image/images/buster/build/mount/
rsync: write failed on "/root/rpi23-gen-image/images/buster/build/mount/var/lib/apt/lists/ftp.debian.org_debian_dists_buster_main_i18n_Translation-en": No space left on device (28)
rsync error: error in file IO (code 11) at receiver.c(393) [receiver=3.1.2]
+ cleanup
+ set +x

cp: cannot stat './images/*.img': No such file or directory

```

quozl, 2018-05-02:
Filesystem size is estimated, but not iteratively proven.  Try increasing the 25% overhead in the definition of ROOT_SECTORS in rpi23-gen-image.sh and repeat the test.

JRWR, 2018-05-02:
Thats what I had to do, I set it to 80% for good measure

JRWR, 2018-05-04:
so I suggest a little more buffer room, 80% sets it to about 500MB for Buster

drtyhlpr, 2018-09-27:
should be fixed

closed 2018-09-27 by drtyhlpr