drtyhlpr/rpi23-gen-image #132: Slipstream Firmware

JRWR, 2018-05-04:
I want to slipstream the brcmfmac43455-sdio into my builds, whats the best way to include that in my rpi3 images

drtyhlpr, 2018-09-28:
Hi, brcmfmac43455-sdio is part of the package: firmware-brcm80211

https://packages.debian.org/stretch/firmware-brcm80211
https://packages.debian.org/buster/firmware-brcm80211

adding the package to APT_INCLUDES="" or APT_INCLUDES_LATE="" parameters should work fine.

burnbabyburn, 2018-10-25:
brcmfmac43455 => RPI3B+

@drtyhlpr 
firmware-brcm80211 is incomplete
see: https://github.com/drtyhlpr/rpi23-gen-image/issues/137#issuecomment-425719949
and my answer/pull req on how to fix it

drtyhlpr, 2018-10-28:
thanks for your great support burn++

closed 2018-10-28 by drtyhlpr