drtyhlpr/rpi23-gen-image #133: Debian (Armel) on the Pi Zero?

cajuncoder, 2018-05-24:
This is not an issue in that it's a bug, but I would love to be able to run straight debian (armel) on the raspberry pi Zero.

Would this be possible? What would be involved? I've read over your project and attempted configuring it to do this with no success yet. When and if I have more time I might look into this more, as I don't quite have the knowledge and understanding yet. For now, this falls somewhere between a feature request and merely opening a topic for discussion (which, as far as I'm aware, github doesn't really facilitate?).

In any case, great project! I hope this is kept going even though debian buster is gaining support for the RPI.

mati75, 2018-05-25:
Devuan (https://devuan.org/) have armel version for rpi 1. It should works on zero too.

burnbabyburn, 2018-11-02:
you could try 

RPI_MODEL=0
KERNEL_ARCH=arm
RELEASE_ARCH=armel
KERNEL_DEFCONFIG=bcmrpi_defconfig
CROSS_COMPILE=arm-linux-gnueabi-

drtyhlpr, 2018-11-26:
should be fixed otherwise please reopen

closed 2018-11-26 by drtyhlpr