drtyhlpr/rpi23-gen-image #136: No rule to make target 'firmware_install' when build kernel

nguyenhungsync, 2018-07-05:
[image]
plz help me

DIKIIDM, 2018-07-16:
I have the same problem

nguyenhungsync, 2018-07-16:
yeah. i fixed .
Solution : 
From kernel 4.14 , i must change in file /bootstrap.d/13-kernel.sh , line 118 

> make -C "${KERNEL_DIR}" ARCH="${KERNEL_ARCH}" CROSS_COMPILE="${CROSS_COMPILE}" INSTALL_FW_PATH=../../../lib firmware_install

**to**  

>  make -C "${KERNEL_DIR}" ARCH="${KERNEL_ARCH}" CROSS_COMPILE="${CROSS_COMPILE}" INSTALL_FW_PATH=../../../lib

drtyhlpr, 2018-08-04:
thanks for the report and quick fix. i am low on time. i would fix this to check/grep if the firmware_install target/string exists in the Makefile. I currently dont have a debian or rpi availible dont know when i will find time to fix this.

drtyhlpr, 2018-09-27:
hi this should be fixed now:

https://github.com/drtyhlpr/rpi23-gen-image/commit/e96075d9ada5e7c209ba96468bbbf12c820ea876

sorry for the delay.

closed 2018-09-27 by drtyhlpr