drtyhlpr/rpi23-gen-image #137: No ethernet on RPI 3 B+ / aarch64

zerkman, 2018-08-05:
Hi,

Thank you for that script which allows to install my favourite system on the raspi.

I successfully built a 64-bit image using the command :

`CONFIG_TEMPLATE=rpi3-stretch-arm64-4.11.y ./rpi23-gen-image.sh
`

Now it boots successfully on my Raspberry 3 B+ (the latest model with 'gigabit' ethernet, among others), but it seems the ethernet controller is not detected. When I look in /proc/net/dev, by default only the 'lo' device appears. After installing the proper binary firmware blobs, wlan0 also appears. But so far I couldn't find any way to activate eth0.

Is there any simple way to have eth0 activated ?

JoeKays, 2018-08-23:
Same here, not pingable, no ssh, keyboard Ctrl+Alt+F1...12 don't do anything.
Seems to be pretty unusable on the 3 B+

g-vidal, 2018-08-27:
I had some issues with latest compilations due to systemd service that was not properly configured by default...

I have added a file `50-wired.network ` in the folder /etc/systemd/network
with  the following contents :

```shell
[Match]
Name=enx*
Host=raspife3

[Network]
DHCP=yes

[DHCP]
RouteMetric=10
```

Then after the first login youcan replace `enx*` by the value given by the instruction `ip token`

Hope it helps...

zerkman, 2018-08-27:
g-vidal,

Thank you for the suggestion, but this did not do anything good.

My opinion is that this is definitely not an issue with basic setup of the network interface. I have done many setups on various RPi's (2,3, and now 3B+), and so far only 3B+ in aarch64 mode has this problem.

g-vidal, 2018-08-28:
Sorry I did not catch that you were using 64bit... I have not yet succeeded to get a raspi functionnal in such configuration...and no time to investigate.
I had also peculiar problems with wicd config files in some builds you can remove and touch  them.

Two dumb questions but sometimes the devil lies in evidences... 

- Before ruling out the basic setup you may also check `/lib/systemd/network/` and check if there is not a debian default parameter  that hurts the Raspi. 
- Have you checked that you have got an operational firmware  for your network chips

finally can you copy what does `ip address`command tells you.

g-vidal, 2018-09-10:
Maybe found a hint!
I have just produced an image 32 bit with **NO DHCP**, I don't know if it is a mistake in my template or a general problem  but _none of the firmware packages_ were installed and of course : no firmware => no driver => network card visible but not active. I added the following line
`APT_INCLUDES="gnupg,firmware-linux-nonfree,firmware-linux,dh-autoreconf,firmware-linux-free"`
I am not sure that all are needed but it solved the problem..... almost!

In fact it seems that at the first boot debian collects informations that are written to /lib/systemd/network/10-eth0.network  and if you use your own `.network` file in `/etc/systemd/network/` and use the `Host` variable with a different name that the original one  there is a conflict and the interface does not come up. I have not looked for the origin of the problem, just change hostname and it worked.

Maybe this could help you trace out the origin of your problem.

zerkman, 2018-09-12:
This is definitely a 64-bit issue. From what I can understand no device tree file for the 3B+ is present in the aarch64 platform files, and the kernel fails to detect the new hardware properly.
Finding the correct files and putting them in the correct place is beyond my knowledge and atm I really do not have time to spend on that subject.

drtyhlpr, 2018-09-30:
https://packages.debian.org/stretch/firmware-linux-free
https://packages.debian.org/stretch/firmware-misc-nonfree

not sure what firmware should be required from this packages. I am currently waiting for my bootstrapping to complete.

Todo: RPi3 wireless should now "simply" work using firmware-brcm80211 , non-free. testing soon

g-vidal, 2018-09-30:
For information.
A colleague reported me a problem with the new rpi3 B+  in 32 bits with wireless.
The new chipset of the pi3B+ is using  `brcmfmac43455` firmware and needs 3 files : 
- brcmfmac43455-sdio.bin
- brcmfmac43455-sdio.clm_blob
- brcmfmac43455-sdio.txt 
but only the first is available in the standard debian package `firmware-brcm80211`. I tried files from [RPi-Distro/firmware-nonfree](https://github.com/RPi-Distro/firmware-nonfree/tree/master/brcmurl) but it solved partially the problem. I ended by cloning files from the raspbian distro which works fine. 
Things will probably be settled down in close future but for the moment there are some complications.

burnbabyburn, 2018-10-25:
in my opinion this is fixed for atleast 64bit with #140 
i applied the same fix as @g-vidal 

dhcp on eth0 is working out of the box
wlan0 device is created by default
i could configure hostaspd to run an AP
i also could configure wpa_supplicant as a client to my network and got a dhcp ip

drtyhlpr, 2018-10-31:
should be fixed. final testing ongoing. feel free to re open

closed 2018-10-31 by drtyhlpr