drtyhlpr/rpi23-gen-image #144: Missing required packages when compiling the kernel

BrainStone, 2018-11-06:
I've noticed a few packages missing from the dependencies list on the readme file.  
While these packages are not needed when not compiling the kernel, they are needed when compiling the kernel.

I think checking for them when compiling the kernel would be appropriate. Also mentioning them in the readme would be nice.

These packages are required when compiling the kernel but missing from the list:
- git (you can get this repo without git. But since it's cloning a repo, git is required)
- make (installed on most systems but not all)
- gcc (again, not always installed) (might also need g++, but not sure)

burnbabyburn, 2018-11-06:
Hi BrainStone,
i've already added some more info on the build system in my fork. I'll add it to my next pull req.
You're sure you read the whole readme? `git` is listed there already. gcc most of the time comes from meta packes below:

Raspberry PI 0 or Raspberry Pi 1 Images, use an ARM (armel) cross-compiler toolchain. (crossbuild-essential-armel)

Raspberry PI 1 or Raspberry Pi 2 Images, use an ARMv7 (armhf) cross-compiler toolchain. (crossbuild-essential-armhf)

Raspberry PI 1 or Raspberry Pi 2 Images, use armhf in 32bit mode and an ARMv8 (arm64) cross-compiler toolchain in 64-bit mode. (crossbuild-essential-arm64)

BrainStone, 2018-11-06:
Oh my bad. I missed the part about the `crossbuild-essential-xxx` and just saw git too. That's naturally also include make etc...  
Sometimes I'm blind I swear.

closed 2018-11-06 by BrainStone