drtyhlpr/rpi23-gen-image #145: iptables

burnbabyburn, 2018-11-06:
The ruleset seems outdated or iptables can not find conntrack module

root@pihole:~# iptables -A INPUT -m conntrack --ctstate RELATED,ESTABLISHED -j ACCEPT
iptables v1.8.1 (nf_tables): Couldn't load match `conntrack':No such file or directory

Try `iptables -h' or 'iptables --help' for more information.

root@pihole:~# lsmod
Module                  Size  Used by
brcmfmac              253952  0
brcmutil               16384  1 brcmfmac
sha256_generic         20480  0
cfg80211              733184  1 brcmfmac
rfkill                 32768  2 cfg80211
cdc_ether              16384  0
evdev                  24576  1
r8152                  65536  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
nfnetlink              20480  0
spi_bcm2835            20480  0
i2c_dev                20480  0
i2c_bcm2708            16384  0
snd_bcm2835            36864  0
vchiq                 331776  1 snd_bcm2835
snd_pcm               131072  1 snd_bcm2835
snd_timer              40960  1 snd_pcm
snd                    90112  3 snd_bcm2835,snd_timer,snd_pcm
ip_tables              28672  0
x_tables               45056  1 ip_tables
ipv6                  507904  26
i2c_bcm2835            16384  0
sdhci_iproc            16384  0

burnbabyburn, 2018-11-07:
seems heavily broken, yet iptables and iptables-persitent still wroking hm...
```
helge@pihole:~$ lsmod
Module                  Size  Used by
brcmfmac              253952  0
brcmutil               16384  1 brcmfmac
sha256_generic         20480  0
cfg80211              733184  1 brcmfmac
rfkill                 32768  2 cfg80211
cdc_ether              16384  0
evdev                  24576  1
r8152                  65536  0
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
nfnetlink              20480  0
spi_bcm2835            20480  0
i2c_dev                20480  0
i2c_bcm2708            16384  0
snd_bcm2835            36864  0
vchiq                 331776  1 snd_bcm2835
snd_pcm               131072  1 snd_bcm2835
snd_timer              40960  1 snd_pcm
snd                    90112  3 snd_bcm2835,snd_timer,snd_pcm
ip_tables              28672  0
x_tables               45056  1 ip_tables
ipv6                  507904  26
sdhci_iproc            16384  0
i2c_bcm2835            16384  0
```
webmin doesnt find iptables

```
An error occured when checking your current IPtables configuration :

iptables: Invalid argument. Run `dmesg' for more information.

This may indicate that your kernel does not support IPtables.
```

burnbabyburn, 2018-11-07:
Okay
1. Debian changed iptables into iptables-legacy and iptables-nf
https://github.com/moby/moby/issues/38099
we need to adopt that and update current iptables ruleset

burnbabyburn, 2018-11-07:
```
custom.d/iptables-legacy.sh 
rm ${ETC_DIR}/alternatives/iptables-restore
rm ${ETC_DIR}/alternatives/iptables-save
rm ${ETC_DIR}/alternatives/iptables
ln -s ${R}/usr/sbin/iptables-legacy-restore ${ETC_DIR}/alternatives/iptables-restore
ln -s ${R}/usr/sbin/iptables-legacy-save ${ETC_DIR}/alternatives/iptables-save
ln -s ${R}/usr/sbin/iptables-legacy ${ETC_DIR}/alternatives/iptables

```

burnbabyburn, 2018-11-08:
IPTABLES FIX is now included in pull request #143 

-set iptables-legacy as default
- add 5s delay to service start to supress "another app is currently holding the xtables lock" error

drtyhlpr, 2018-11-09:
hi burn. wow thanks for your report and all your fixes. good job!

I am a little bit frustrated with this "nice little changes" (debian/kernel/systemd) that break this bootstrapping script aggain and aggain and if you can only find a small statement about a change in a maillinglist post or change log. (cryptsetup f.e.)

burnbabyburn, 2018-11-09:
Yeah that's not very funny at all!
Most importantly: You HAVE to feel happy if you find an information snippet telling you what they broke (again) and how! Especially if systemd is involved... Trying to find infos other than the man page is hard. 

Another example: Debian guys killed hostapd (masked the service) in favor for systemd. Good luck finding information in how to do an access point in/with systemd :D

burnbabyburn, 2018-11-19:
fixed

closed 2018-11-19 by burnbabyburn